DarkFlippers/unleashed-firmware #361: Infrared keyboard

xfmxx, 2023-02-23:
### Describe the bug.

When I try to save a new remote(button) the keyboard letters are capital so they should write capital letters right? But when I type it's small. When I save the remote and remove the example text, it's an capital letter. 

https://user-images.githubusercontent.com/78693466/220982738-d44c584e-dedc-4f19-a739-1bee48f82256.mp4

Also normaly when theres a keyboard left from the save button theres a `_`. when it's an capiltal keyboard it also shows "_" but writes an space like when you longpress the `_` in normal mode. 
The funny thing is when theres an example text and the keyboard should write capital letters the `_` wents to an ` ` but also writes `_`.

https://user-images.githubusercontent.com/78693466/220982717-1112adf8-8d0c-47c2-a37b-3f2d50c5dfcd.mp4

### Reproduction

Described above

ClaraCrazy, 2023-02-23:
I'm curious, can you replicate this anywhere else? I remember having had such an issue on my Firmware too, but I eventually fixed it. I'm on a trip right now but if you can confirm this to be present everywhere I know its my old issue and I can PR a fix once im back home

xfmxx, 2023-02-23:
I tried it on SubGHz and theres the same problem.

ClaraCrazy, 2023-02-23:
Good enough for me. I'll PR a fix in the coming days. 

Or actually… @Willy-JL can you do that? Remember when I did that tiny change on the keyboard input.c? That's what's needed here.

Willy-JL, 2023-02-23:
Will do :ok_hand:

PR #362 (2023-02-24): Fix keyboard capitalization bug
# What's new

- Fix keyboard capitalization
- Closes #361 

# Verification 

- Properly types uppercase at beginning of textbox
- Keyboard keys and typed characters are coherent

# Checklist (For Reviewer)

- [x] PR has description of feature/bug
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2023-02-24 by xMasterX