DarkFlippers/unleashed-firmware #364: Could a option to turn on infrared debug pin be added to infrared menu?

bushbooger, 2023-02-24:
### Description of the feature you're suggesting.

option On/Off  to send IR signal to debug pin
To be used for high-power IR blaster hats  

### Anything else?

Thank you everyone for making flipper awesome!

xMasterX, 2023-02-26:
Done, in dev branch

closed 2023-02-26 by xMasterX

xMasterX, 2023-02-26:
Also will be in 032 release

nullvoid413, 2023-02-28:
Any docs on this?

1. Where the pins should connect
2. If it works with transmit and receive
3. Will external IR modules work or if it only uses the internal module with the ability to use an external diode led..?

Thanks

xMasterX, 2023-02-28:
> Any docs on this?
> 
> 1. Where the pins should connect
> 2. If it works with transmit and receive
> 3. Will external IR modules work or if it only uses the internal module with the ability to use an external diode led..?
> 
> Thanks

1 - It literally says pin number in settings - 2(A7) + GND
2 - Only transmit
3 - Only one module can work, internal or external if enabled