DarkFlippers/unleashed-firmware #382: Sub-ghz

Antonzab75, 2023-03-12:
### Description of the feature you're suggesting.

Good afternoon!!!  is it possible to add: 1. Sub-ghz display of the serial number of the remote control in the rolling code, as in the drain.  2. in the rolling code to the doorhan button programming (long press button 2 and then simultaneously button 1).  Thank you in advance!!!

xMasterX, 2023-03-12:
Already added, looks like you haven't tried pressing different buttons (left, right, up, down) in saved doorhan remotes
Serial number is displayed as FIX, first digit is a button, all other data is actual serial

closed 2023-03-12 by xMasterX

Antonzab75, 2023-03-12:
I know about the buttons, but they can't be pressed at the same time.

xMasterX, 2023-03-12:
> I know about the buttons, but they can't be pressed at the same time.

you don't need to press two buttons at same time

button 0xA will act as 1+2 on doorhan

Antonzab75, 2023-03-12:
To program the Doorhan remote control, first press button 2 and after 2 seconds button 1, how can this be done?