DarkFlippers/unleashed-firmware #402: Sub-Ghz auto mode write new key receiver

TheMaxStr, 2023-03-27:
### Description of the feature you're suggesting.

Add a master key selection function - an opening gate and a key slave - a manually created key in order to automatically register new keys in the receiver

xMasterX, 2023-03-28:
What do you mean? what gate? what protocol it uses?

TheMaxStr, 2023-03-29:
Gate nice/came. In order to register a new key, it is necessary to do a number of manipulations with 2 keys and maintaining certain timings of pressing the buttons (nice: long press 5s, work key 3 press by 1s and long press 2s new key). I suggest automating this procedure. By selecting both keys one that opens the other that was added manually and that needs to be added

xMasterX, 2023-03-29:
First of all, CAME and Nice are different brands and manufacturers
Second - you didn't said anything about model, and what protocol it uses, how much times i should say that?
I can't guess by looking on your message, you described some programming procedure, no info about exact system and model is provided

TheMaxStr, 2023-03-29:
I'm talking about the key programming mode. I describe the case: An application in which you can select two keys *.sub, the first file with the master key that opens the barrier, the second file with the key that you need to register in the receiver. After selecting these 2 files, you can press the start button and the flipper itself will perform all the manipulations that are necessary for remote prescribing, instead of counting these seconds by itself. Do you understand?

xMasterX, 2023-03-29:
i'm not doing that, ill keep this issue open in case someone who has similar issue with opening two files at such short period of time, maybe they will make an special app for your case
also, why you don't have master remote, why it is in flipper, where is actual remote? you stolen someone's signal and trying to bind your remote using that?

TheMaxStr, 2023-03-30:
I'm talking about the possibility that is laid down by the manufacturer of these barriers and only about those use cases that this manufacturer implied by laying the possibility of remote programming. Nothing prevents you from having the originals of the remotes in your hands to make copies of them in flipper and insert into the applications described by me as a master and slave key, in order to actually simplify this programming process. All other cases remain outside the brackets and on the conscience of those who use it for selfish purposes

billy20301202, 2023-04-04:
I encountered a remote control with a frequency of 432.8

nahaki, 2023-06-14:
Thanks for sharing