DarkFlippers/unleashed-firmware #411: Add support to rc "multi" modules (rc car and drone remote control)

mankalan, 2023-04-05:
### Description of the feature you're suggesting.

Hi!
I was wondering if someone can add support for multi protocol https://github.com/pascallanger/DIY-Multiprotocol-TX-Module
This is a module, based on nrf24 and if i remember well you can made it or buy it in aliexpress, that module can conect to almost any drone/rc car toy and control it. i think its simple add this as an app becouse it comunicates via serial interface and ppm (a radio control protocol) and I think this will be very cool to add to a flipper, add like basic operations, turn on or off, maybe a basic control for rc cars... but i dont have the knolwedge to program that kind of software, so i post it here to someone pick my dream and make it true ;)

fzarada, 2023-04-07:
I believe it's called Sub-GHz in flipper for a reason

mankalan, 2023-04-19:
> I believe it's called Sub-GHz in flipper for a reason

Okay so you clearly didn’t read the whole post.
This module I’m talking about are the transmitters! So, you don’t need to use the subghz module of the flipper for anything. Also, this module takes the ppm signal you send to it, and translate to the rc dron/car you choose. Another option I was talking is to make this more accessible to people and if it goes viral don’t break the AliExpress stock, just using a nrf24 module (another transmitter module) with the flipper and add the official code of the Moule (opensource by the way) and implement it on the flipper to run as an app. The code is ready to talk to the nrf24.

Also, maybe if this go well, I can design a pcb for attach to the flipper that have joysticks in it to make it even more useful and cool. Or maybe, idk we can connect a ps4/ps3/Xbox controller to the flipper via Bluetooth and translate the controls to the multiprotocol module. 
This is something that with an Arduino like board I can make very easy, but I’m not a programmer, and also, I don’t know how to program in the flipper, that’s  why I was asking for help, because I think this is a pretty good idea and also, I have electronics knowledge so collaborating with other developers we can make this a real thing.
So please, read the whole post before you say anything.