DarkFlippers/unleashed-firmware #413: Picopass app not reading elite keyed cards anymore.

Akury83, 2023-04-07:
Hi all,

I just upgraded from ver 034 to ver 042 and I’ve found that the picopass app no longer reads elite keyed cards, even ones keyed with the keys in dictionary files. The old version of the app used to read elite keyed cards with the default or icopy-x keys..

Any help would be greatly appreciated with this one, cheers!

Akury83, 2023-04-15:
Hello all, tested with the new firmware: 043, still not working…

xMasterX, 2023-04-16:
Can you tell more details? We are not making this plugin, it came from OFW and made by @ bettse 
What you doing to read it?
App has added new menu option, to read elite cards, do you using it or trying default read?

Akury83, 2023-04-16:
Ohh I see, I had no idea it was just pulled from the OFW.. Thanks for letting me know.

In the old version, it would read standard or elite keyed cards with the one read option.

In the new version, it reads standard cards with the regular read option, but doesn’t read elite keyed cards using the ‘Elite dict attack’ option, even after going through the dictionary of available keys.

Considering this is pulled from the OFW, should I close this request and raise the request there?

xMasterX, 2023-04-16:
Correct, also better to try it by installing latest OFW dev build first, to make sure issue still present
Also, You tried both read and elite dict options on that card? 
None of them worked?

Akury83, 2023-04-16:
Tried the latex OFW, and OFW dev build, both failed to read an elite coded card using both elite and elite dict options. The card is coded in the standard key, but with the elite algorithm. Thanks for your help, I’ll close this and raise in the official channels.

closed 2023-04-16 by Akury83