DarkFlippers/unleashed-firmware #414: Internal Storage Blockage on self-built .dfu Installation

lesterrry, 2023-04-10:
### Describe the bug.

## What I am trying to do:
- Clone this repo
- Run `./fbt`
- Install the built .dfu file via qFlipper

## What I expect:
- Firmware being flashed and installed on the device with no further issues

## What I get:
- Firmware being flashed and installed on the device 
- 0 KiB free displayed in _About Internal Storage_ tab in device settings
- No adjustments to device's settings (e.g. adding pin code) may be made (they just won't get applied)
- To restore the internal storage repairing the device is required through qFlipper, installing in the process default (not unleashed) release version of firmware

## What I've also tried:
- Updating via Updater Package from SD Card
- Updating via built .tgz file

## What I will not try:
- Update via Web Updater or any other way through pre-built releases. I am **specifically** looking to build the firmware myself

### Reproduction

- Clone this repo
- Run `./fbt`
- Install the built .dfu file via qFlipper

lesterrry, 2023-04-11:
> Why do you using debug builds then? build release

I do need debug build for development, so I used the instructions given in ['How to build'](https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/HowToBuild.md). 

Building debug version with `./fbt FIRMWARE_APP_SET=debug_pack updater_package` did work. Thank you! 

As simple `./fbt` command given in ['How to build'](https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/HowToBuild.md) does not work, I suggest you update the documentation on building the firmware according to this issue. I will create a PR with updates soon.

xMasterX, 2023-04-13:
Docs updated, thanks for information

closed 2023-04-13 by xMasterX