DarkFlippers/unleashed-firmware #422: Getting stuck in walls playing Doom

trigat, 2023-04-15:
### Describe the bug.

I'm not sure if I should post this here or in another Doom repository.  I updated my firmware to Unleashed yesterday and I notice I always get stuck in the walls when playing Doom.  Is this something that can be looked at?  It happens any time I touch the walls.

### Reproduction

Play Doom and run into walls.

xMasterX, 2023-04-16:
Good find, this bug is almost year old, I have no time to fix it, and no one reported it since then
Fixing DOOM is lowest priority for me, since its not a proper game, its more like a demo, and it works good as a demo

trigat, 2023-04-16:
Good to know.  Thank you.

closed 2023-04-16 by trigat