DarkFlippers/unleashed-firmware #427: Add *9bit 318MHz* to Applications>Sub-GHz>Sub-GHz Bruteforcer

hotdjdave, 2023-04-18:
### Describe the enhancement you're suggesting.

I have several remotes (in the USA) with 9-pin dip switches (9bit) to open gates/garages, and they all transmit on 318MHz.  I'm not sure if it is Chamberlain, Linear, other, or even if that matters.  Could this frequency be added as one of the options under Applications > Sub-GHz > Sub-GHz Bruteforcer?

Thanks.

### Anything else?

Also, as a side note, none of the _deBruijn_ applications seem to work.  I've tested all of them on gates where I know the frequency and dip-switch settings.

hotdjdave, 2023-04-30:
There are two different remotes:  a larger gray one and a smaller whitish one with a black button.  If you need anything else please let me know.  Thanks.  David. 
[image]
[image]
[image]
[image]
[image]
[image]

hotdjdave, 2023-04-30:
[image]

hotdjdave, 2023-04-30:
Uploading 20230429_231537.mp4…

hotdjdave, 2023-04-30:
https://user-images.githubusercontent.com/130207130/235338851-1e48e864-8404-44ef-b184-d7d320be67cb.mp4

Videos won't upload.

hotdjdave, 2023-04-30:
https://user-images.githubusercontent.com/130207130/235338952-f649563e-ba3d-41ad-a3e4-3b4bcd6be979.mp4

hotdjdave, 2023-04-30:
[image]

hotdjdave, 2023-04-30:
[image]
[image]

hotdjdave, 2023-05-08:
I think I meant:  _CAME_brute_force_, but wherever it may be needed.

hotdjdave, 2023-05-20:
Is anyone working this?  Thamks.

hotdjdave, 2023-05-30:
Any progress?

xMasterX, 2023-05-31:
> Any progress?

Why you keep asking for something again and again, you still not sent RAW recordings I asked about in my first message

hotdjdave, 2023-05-31:
> > Any progress?
> 
> Why you keep asking for something again and again, you still not sent RAW recordings I asked about in my first message You think its possible to do something by looking on photos?

Sorry, I uploaded at least two videos to this thread.  I was not sure if anyone had seen this.  If the you are unable to see the videos or if the information in the videos is not sufficient, please let me know what to do.

Thank you.

xMasterX, 2023-05-31:
> Sorry, I uploaded at least two videos to this thread. I was not sure if anyone had seen this. If the you are unable to see the videos or if the information in the videos is not sufficient, please [...]

are you joking? on first two videos you doing exactly what i asked for
I said:
> Provide raw recordings of multiple presses of button on your remote

Where you sent me that files? There is none, only video showing how you recording them

hotdjdave, 2023-06-01:
> Provide raw recordings of multiple presses of button on your remote, and photos + model / fccid

I'm sorry.  I misunderstood you.  I thought when you wrote "raw recordings" you meant video recordings.  That is how I understood it.  I will send the files of the recordings I record on F0.

hotdjdave, 2023-06-01:
[F0 Raw Sub Files for Bruteforcer.zip](https://github.com/DarkFlippers/unleashed-firmware/files/11628272/F0.Raw.Sub.Files.for.Bruteforcer.zip)
Files are attached.  Github won't let me upload .sub files, so I zipped them.  The one that ends in AS is for the Allstar remote and the one that ends in MC is for the Multi-Code remote.  Let me know if this works for you. Thank you very much.

xMasterX, 2023-06-01:
> [F0 Raw Sub Files for Bruteforcer.zip](https://github.com/DarkFlippers/unleashed-firmware/files/11628272/F0.Raw.Sub.Files.for.Bruteforcer.zip) Files are attached. Github won't let me upload .sub [...]

Good, will take a look on them later, also due to lack of flash space adding support of those protocols might be delayed for some time (there is no reason to add them only in bruteforcer app, better will be to add them at subghz lib level)