DarkFlippers/unleashed-firmware #444: Reading RAW RFID File error when emulating RAW RFID

alexle-dev, 2023-04-28:
### Describe the bug.

When I try to emulate a RAW RFID file, I get the following error: `Reading RAW RFID File error`.
I'm using `unlshd-046`.

### Reproduction

1. Switch on
2. Go to 125kHz RFID
3. Go to Extra Actions
4. Select Read RAW RFID data
5. Scan your RFID tag
6. Go back and select Emulate RAW RFID Data
7. Try emulating your RFID tag

xMasterX, 2023-05-22:
Can't replicate, send your raw files

Leptopt1los, 2024-01-06:
We are closing the issue because you did not provide files for testing. if the problem is reproduced on the latest firmware version - reopen this issue

closed 2024-01-06 by Leptopt1los