DarkFlippers/unleashed-firmware #449: FlipperZero crashed while standby

UrSuLa360, 2023-05-01:
### Describe the bug.

Unleashed version **046**
I leave flipper zero with screen locked and after a while i found **"Flipper crashed and was rebooted
NULL pointer dereference"** on screen.

I also found **"Flipper crashed and was rebooted
furi_check failed"** message

### Reproduction

Flipper with Unleashed version 046
Linked to Flipper iOS app via bluetooth
Just lock the screen and leave it.
The error will appear (you need to wait some hours, i found it three times in three days)

### Target

os

### Logs

```Text
Some logs i could recover:
Log 1
[23:24:45][error]: COUNTLY_APP_KEY not found
[23:24:45][info]: session started
[23:24:49][info]: canceling tasks...
[23:24:49][info]: canceling tasks done
[23:24:49][info]: session started
Log 2
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][debug]: widget connection time is out
[20:42:09][info]: session started
[20:42:10][info]: canceling tasks...
[20:42:10][info]: canceling tasks done
```

UrSuLa360, 2023-05-04:
Still happening in 047e.

A new error message appears "**Flipper crashed and was rebooted MPU fault, possibly stack overflow**"

UrSuLa360, 2023-05-05:
The bug doesn't happen if you change the sleep method for FlipperZero to legacy.

_Settings-> System ->Sleep Method_ = _Legacy_ (No crash)
_Settings-> System ->Sleep Method_ = _Default_ (will crash)

UrSuLa360, 2023-05-10:
Still happening in 048e.

Only me having this bug? 😓

TheWebMachine, 2023-05-10:
I haven't had this problem at all. 🤷🏻‍♂️ Maybe you just have a faulty unit?

T3CHKOMMIE, 2023-05-10:
48e here. I've had the idle crash the last two days

K-Dog98, 2023-05-10:
When I try uploading, the new firmware unleashed to my Flipper zero it gets to about 70% and then says couldn't connect to Flipper. 
Has anyone got some ideas on what to be going on?

xMasterX, 2023-05-12:
This is a known issue that happens only on Some flippers, not at all units, no, its not a hardware fault, issue comes from core 2, and can be fixed in software, official team is working on solution, also its very hard to debug since core2 cannot be debugged, debug interface is not accessible, and its firmware is encrypted and closed, we can use binary files provided by manufacturer

FoxLy1312, 2023-05-12:
Do a factory reset first, mostly if i got this problem internal storage is full

FoxLy1312, 2023-05-12:
Use a short cable like the original one

xMasterX, 2023-05-18:
As temporarily solution, you can try using this - Settings -> System -> Sleep Method = Legacy
Battery life will be as on older releases
We can't say when fix will be ready, since it depends from ST, chip manufacturer, core2 runs closed encrypted firmware made by them

Also as we find out, only some flippers are affected, but there is nothing similar between them, revision, manufacturing date, this is fully random
So if you have 2 flippers and only one of them crashes, that doesn't mean its chip revision or manufacturing defect, core2 happens to act a bit different on some devices which has nothing similar between them
Current solution -> Use it with `Sleep Method = Legacy` and wait for next updates

UrSuLa360, 2023-06-13:
> Core2 (Crash in idle) issues was reduced to current possible minimum, you can try using DeepSleep again (Sleep Method = Default) (+ more checks was added, if you get Slow HSE/PLL startup message more than one time, create issue with steps what to do to reproduce it again)

Still happening in 053.
Anything you need to help let me know

niko-lay, 2023-06-14:
I have faced same issue with 052 firmware, flipper connected via Bluetooth to ios (looks like it start doing auto-connect few versions ago) and stay locked for few hours. Also sleep method=default
Error message on screen: "Flipper crashed and was rebooted. STR(R) Copro(R) HardFault"

niko-lay, 2023-06-14:
> I have faced same issue with 052 firmware, flipper connected via Bluetooth to ios (looks like it start doing auto-connect few versions ago) and stay locked for few hours. Also sleep method=default Error message on screen: "Flipper crashed and was rebooted. STR(R) Copro(R) HardFault"

just googled few and found this is known issue with 
https://forum.flipperzero.one/t/message-on-screen-flipper-crashed-and-was-rebooted-st-r-copro-r-hardfault/16948/2

xMasterX, 2023-10-10:
Test it with latest 064 release, should be fixed now

UrSuLa360, 2023-10-11:
Testing for next 2 days, if no error found will close it. Thank you @xMasterX for your work

If anyone want to test this issue, please **dont forget** this steps:
**1. Update flipper with 064 of unleashed version
2. Settings-> System ->Sleep Method = Default**

UrSuLa360, 2023-10-13:
No trace of the error during last 2 days on my unit. Closing the issue.

closed 2023-10-13 by UrSuLa360