DarkFlippers/unleashed-firmware #457: Extern C1101 dosent work anymore

oldbug62, 2023-05-06:
### Describe the bug.

My external C1101 modul dosent work anymore flipper freze when i turn to external. anybody have the same issue?

unlshd-047

### Reproduction

put c1101 at gpio
switch sub-ghz
flipper freze

other moduls work (nrf24)

awnyel, 2023-05-06:
I noticed same issue it only catches few buttons from few of the remotes

krolchonok, 2023-05-07:
What module do you have? DIY? Or a purchased fee?

xMasterX, 2023-05-07:
We tested external radio many times, and your issue is not related to firmware at all

Issue comes from bad contact or broken module
If you made it yourself, check your connections, and clean up flux after soldering, this causes issues too

Latest firmware can't freeze in subghz read, if module is working properly and connected properly, so check your module and its connections, try different one

closed 2023-05-07 by xMasterX

awnyel, 2023-05-07:
it's
 just impossible since all was working perfectly and I tested on a few boards and cc modules also different keyfobs with the same results.

krolchonok, 2023-05-07:
> it's
>  just impossible since all was working perfectly and I tested on a few boards and cc modules also different keyfobs with the same results. 

on which version does everything work for you?

awnyel, 2023-05-07:
i just updated yesterday now i tried to downgrade to get them back without luck. and this problem is across all fw with latest updates.

awnyel, 2023-05-07:
and it worked like a month ago

oldbug62, 2023-05-07:
@awnyel  exactly the same here