DarkFlippers/unleashed-firmware #459: Stealth mode should disable vibration motor

alexl83, 2023-05-08:
### Describe the enhancement you're suggesting.

I think it could be useful to disable all kind of obvious outputs in stealth mode
Silencing sound is great, and I think disabling vibration would enhance the functionality

While running covert, vibration sound could more or less easily give away some activity and raise suspicion

### Anything else?

Whishlist-level suggestion
Thank you for your consideration - thank you for this great job with unleashed firware

alexl83, 2023-05-10:
UPDATE: checked subghz READ and subghz "frequency analyzer" - Stealth mode disables vibration already, still unsure about behaviour in other apps - led blinking still active

tested on UNLSHD-48 release
Thank you

xMasterX, 2023-05-22:
Stealth mode disables vibro and sound as is should

closed 2023-05-22 by xMasterX