DarkFlippers/unleashed-firmware #460: "Test" missing in Sub-GHz after unlshd-048 update

hotdjdave, 2023-05-08:
### Describe the bug.

I recently updated to unlshd-048.  I noticed the "Test" feature under Sub-GHz is now missing.  I used to come right after "Radio Settings."

I'd show a screenshot, but this form won't let me.

### Reproduction

1. Go to Sub-GHz menu.
2. Scroll through menu items and you will see "Test" is missing.

hotdjdave, 2023-05-08:
Additional information.   In using F0 to open a gate using _Sub-GHz CAME_brute_force_ works, but not only via internal antenna.  Using the external antenna with my attached CC1101 module used to work fine, but now does not.  I'm not certain, but it the issue seems to correspond with the update.

ALD53105, 2023-05-09:
Looks like test for Sub-Ghz has been moved to the debug menus. Settings > System > Debug > On, you'll then have test back.

xMasterX, 2023-05-09:
> Looks like test for Sub-Ghz has been moved to the debug menus. Settings > System > Debug > On, you'll then have test back.

It was hidden if debug disabled all the time from the start, nothing was changed, looks like someone was using flipper with debug = on, 24/7

closed 2023-05-09 by xMasterX

hotdjdave, 2023-05-09:
> 

Is it bad to leave Debug on?

hotdjdave, 2023-05-09:
> > Looks like test for Sub-Ghz has been moved to the debug menus. Settings > System > Debug > On, you'll then have test back.
> 
> It was hidden if debug disabled all the time from the start, nothing was changed, looks like someone was using flipper with debug = on, 24/7

Is it bad to leave debug on?