DarkFlippers/unleashed-firmware #466: Saved NFC - cannot parse file

BoardCZ, 2023-05-14:
### Describe the bug.

When I read card via NFC (RFID and ibutton works) and safe it. I cannot emulate it, because I always get error ‘cannot parse file’. 
I already try another new SD card.
When I download saved file and open it with text editor, it contains data.

### Reproduction

1. read card with NFC
2. save if
3. try to emulate it

xMasterX, 2023-05-14:
send your file here

BoardCZ, 2023-05-16:
Here is the file from Flipper. I changed file name extension, because GitHub not allowed upload file with extension .nfc
[Nejde.txt](https://github.com/DarkFlippers/unleashed-firmware/files/11484870/Nejde.txt)