DarkFlippers/unleashed-firmware #469: Security+2.0 remote not working

victor-s-yip, 2023-05-16:
### Describe the bug.

On a craftsman overhead garage door, the flipper can register rolling security+2.0 at 310 Mhz and 390 Mhz. When using the registered code, the garage unit indicator appears to blink as if it read something but will not open or close. Running 048e. 

### Reproduction

1) Make a new security+2.0 rolling code remote at 310/390Mhz
2) Activate "learning" on unit
3) continuous transmit until receiver blinks confirmed (as per manual)
4) Try to transmit again to open/close door.

xMasterX, 2023-05-16:
> Sorry about that. I was not sure if this issue was more widespread because it was referenced elsewhere without resolution. The original remote is recognized as a security+2.0 at 390Mhz. Would a normal read or a raw read be preferred for this?

Send both RAW and signal you got from regular read

victor-s-yip, 2023-05-27:
Attached is a .zip with the Raw and read (315 Mhz) for the working remote as well as copies of the "Add manually" that were able to register successfully, but not operate the door, 310, 315, and 390 Mhz.

[Remote_files.zip](https://github.com/DarkFlippers/unleashed-firmware/files/11583109/Remote_files.zip)

xMasterX, 2023-05-28:
> Attached is a .zip with the Raw and read (315 Mhz) for the working remote as well as copies of the "Add manually" that were able to register successfully, but not operate the door, 310, 315, and 390 [...]

Good, thanks, I found an issue and fixed that, different receivers uses different button codes, I added some of them, but yours was not in the list, now its supported
Install release 050 when it will be published (soon), and try this: Add manually - SecPlus 2.0, Then hold Right arrow button in emulate screen to emulate button with `0x78` code like in your original remote, bind that remote to receiver using only that button, all should work

closed 2023-05-28 by xMasterX