DarkFlippers/unleashed-firmware #472: MemManage crash when usb disconnected

slyker001, 2023-05-17:
### Describe the bug.

Got MemManage crash after updating (not first time)
When usb disconnected flipper starts rebooting, if debug is turned on it stops crashing (some times connecting to powerbank also stops it, but until next usb connection)
if it matters pc connection via type-c type-c cable with usb A adapter

https://youtu.be/xpDCZVqcrJo

### Reproduction

1. Install via web updater
2. Disconnect USB cable

xMasterX, 2023-05-17:
Why do you have Heap Trace enabled?

slyker001, 2023-05-18:
idk, without it still crashing. i think i enable it when it first happened and i tried to get any logs or something i can attach to issue
without bluetooth still crashing
eject sd card still crashing
with legacy sleep method not crashing

i think it started at unleashed version 043 or 045, don't remember this, somewhere around these versions

xMasterX, 2023-05-18:
> idk, without it still crashing. i think i enable it when it first happened and i tried to get any logs or something i can attach to issue without bluetooth still crashing eject sd card still [...]

Legacy sleep method? Yes, your issue is 100% related to #449
You can set it in legacy for now, battery life will be like in older releases, but at least no crashes for you at the moment

Please read #449 thread and my answer in it, you need to wait, ill update that issue when fix is ready, for now Im closing this issue, you can subscribe to #449

closed 2023-05-18 by xMasterX