DarkFlippers/unleashed-firmware #476: Firmware flash error on unlshd-049

coded-with-claws, 2023-05-18:
### Describe the bug.

When I checkout branch release then compile (./fbt && ./fbt fap_dist), then I install the dfu with qflipper, it ends with error, and I have to use the Repair function to restore the original firmware.
I tried three times.
I never had this problem with the previous releases of Unleashed.

### Reproduction

1. git clone https://github.com/DarkFlippers/unleashed-firmware.git
2. cd unleashed-firmware
3. git checkout release
4. ./fbt (no error)
5. ./fbt fap_dist (no error)
6. qflipper, Install from file, select dist/f7-D/flipper-z-f7-full-local.dfu

### Logs

```Text
42208 [DBG] Erasing memory: 96%
42350 [DBG] Erasing memory: 97%
42486 [REG] Device went offline: VID_0x483:PID_0xdf11
43652 [DBG] Failed to perform control transfer: LIBUSB_ERROR_NO_DEVICE
43652 [DBG] Unable to get device status
43652 [DBG] An error has occurred during erase phase
43652 [DBG] Failed to erase page
43652 [DBG] Failed to erase the memory
43653 [RCY] Firmware Download @Urbik ERROR: Can't flash firmware: An error has occurred during the operation.
43653 [DEV] Firmware install from file @Urbik ERROR: Can't flash firmware: An error has occurred during the operation.
43653 [BKD] Current operation finished with error: "Can't flash firmware: An error has occurred during the operation."
85287 [REG] Detected new device: VID_0x483:PID_0xdf11
85313 [DBG] Device reported transfer size: 1024
85313 [DBG] Bytes uploaded: 32 100%
85313 [DBG] Upload has finished.
```

### Anything else?

I hope you reproduce the problem so that it can be fixed 🤞

xMasterX, 2023-05-18:
You doing it wrong, follow instructions:
You compiling debug build which contains all the apps and other stuff, compiling full debug is not possible for a long time
You need to add extra parameter if you want to build debug builds, in your case you want release build, this has different parameters at all, also flashing .dfu means NO ASSETS, so assets, plugins, will not be updated, why you can't use proper .tgz update packages build by fbt?

https://github.com/DarkFlippers/unleashed-firmware/blob/dev/documentation/HowToBuild.md

closed 2023-05-18 by xMasterX

coded-with-claws, 2023-05-18:
Thanks for the info.
I used to compile without parameters since beginning (summer 2022) and flash with .dfu because there were no external apps back then.
I read the doc and compiled in release mode to get the .tgz update package:
`./fbt COMPACT=1 DEBUG=0 updater_package`
This gives me a tgz having the same size than the Unleashed release one (1,7M).
But now in qflipper doing the install, I get a new error:
```
52405 [RPC] (12) Storage Write @/ext/update/f7-update-local/update.fuf START
52470 [RPC] (12) Storage Write @/ext/update/f7-update-local/update.fuf SUCCESS
52470 [RPC] (13) Storage Write @/ext/update/f7-update-local/updater.bin START
53271 [RPC] (13) Storage Write @/ext/update/f7-update-local/updater.bin SUCCESS
53271 [UTL] Upload 6 entries @Urbik SUCCESS
53281 [UTL] Start Update @Urbik START
53283 [RPC] (14) System Update @/ext/update/f7-update-local/update.fuf START
53336 [UTL] Start Update @Urbik ERROR: Internal storage is full
53336 [DEV] Full Update @Urbik ERROR: Internal storage is full
53336 [BKD] Current operation finished with error: "Internal storage is full"
53341 [RPC] (14) System Update @/ext/update/f7-update-local/update.fuf SUCCESS
```
It seems to be a common issue, needing a recovery. I did recoveries this morning before posting this issue, and I also tried to do "Erase" from qflipper, so I don't know.
For the moment, I have managed to flash the .dfu + added manually the apps like I used to do. It worked with the new fbt  command line.