DarkFlippers/unleashed-firmware #488: Issues Setting "Primary Favorite App" in unlshd-050

hotdjdave, 2023-05-28:
### Describe the bug.

Upgraded to **unlshd-050**, now setting Primary Favorite App has issues.  I used to be able to select "Applications" as an option and it would let me go down further in the tree to select and actual app.  However, not it only allows me to chose the first level of the tree:  Application, Sub-GHz, Sub-GHz Remote, 125 kHz RFID, etc., without letting me go deeper into the selection.  

### Reproduction

Select:
1).  Settings
2).  Desktop
3).  Primary Favorite App
4).  Applications (or any other in menu)

hotdjdave, 2023-05-28:
https://github.com/DarkFlippers/unleashed-firmware/assets/130207130/a4e594c1-9ed9-4b2c-b903-7267c35e00aa

xMasterX, 2023-05-28:
Have you tried scrolling to the bottom? Now it says [External app]

closed 2023-05-28 by xMasterX