DarkFlippers/unleashed-firmware #489: Trying to rename some files in Sub-GHz module ends with FW crash

marcinx64, 2023-05-28:
### Describe the bug.

When trying to rename file on SD card from Sub-GHz menu, FW is crashing with BusFault error.

FW: UNLSHD-49E
FW: UNLSHD-50N

### Reproduction

1. Go to Sub-GHz
2. Click "Saved"
3. Pick file (in my case "433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub")
4. Click "More"
5. Choose "Rename"
6. FW crash with BusFault error displayed after reboot

### Anything else?

This issue is not observed on Official FW (0.83.1)

xMasterX, 2023-05-29:
> Click "More"

Where do you see this More menu? There is nothing like that
Also I tried rename in different places, all works ok
Plus I tried files with `.` in their names
And I tried naming file as yours, then renaming it - no issues

Try reinstalling firmware using web updater, or see #449 
This might be related, and using rename was just coincidence

marcinx64, 2023-05-29:
It is reproducible everytime, so clearly not related to #449 imo, but renaming works ok for some files (for example renaming 433_Tesla-Long_AM650_EU_AUS.sub signal works without issues). Here is a video of reproducing and picture after crash:

[Nagranie ekranu z 2023-05-29 12-30-04.webm](https://github.com/DarkFlippers/unleashed-firmware/assets/43246702/1af2e62c-8382-4b21-af23-0ba751342311)

[image]

attached file(please change txt to sub as GH is not allowing sub files here):
[433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.txt](https://github.com/DarkFlippers/unleashed-firmware/files/11591030/433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.txt)

xMasterX, 2023-05-29:
Fixed in dev branch, will be in 051 release

closed 2023-05-29 by xMasterX