DarkFlippers/unleashed-firmware #496: 125 khz - tag from electra intercom

Furrypaw, 2023-06-01:
### Description of the feature you're suggesting.

Add support to electra intercom 125 khz tag.

There is a discussion on the forum Reddit
https://forum.flipperzero.one/t/electra-intercom/6368

There are also seemingly some work done on [Flipper-Xtreme] firmware.
https://github.com/ClaraCrazy/Flipper-Xtreme/issues/199

xMasterX, 2023-06-02:
> ### Description of the feature you're suggesting.
> Add support to electra intercom 125 khz tag.
> 
> There is a discussion on the forum Reddit https://forum.flipperzero.one/t/electra-intercom/6368
> 
> There are also seemingly some work done on [Flipper-Xtreme] firmware. [ClaraCrazy/Flipper-Xtreme#199](https://github.com/ClaraCrazy/Flipper-Xtreme/issues/199)
> 
> ### Anything else?
> _No response_

What work was done in XFW? I don't see anything except moving issue to "TODO"

As for protocol support itself, i see some more work has been done by [dcaprita](https://forum.flipperzero.one/u/dcaprita)

I will take a look how to remake it in a proper way

senciucserban, 2023-06-20:
Looking forward for this. Is there any way to support your work on this?

xMasterX, 2024-06-14:
Hello,
Full protocol implementation was made
 https://github.com/DarkFlippers/unleashed-firmware/pull/750

Now you can use it starting from 073 release and every next one

closed 2024-06-14 by xMasterX