DarkFlippers/unleashed-firmware #497: Deauther wont work esp8266 board only

DavidIgor, 2023-06-02:
### Describe the bug.

deauther in apps is not working , flipper crashes and restarts each time , when 5v turned on in GPIO it just freezes on loading screen .  fw 50 and esp8266 flashed via browser as per instruction  !
deauther was working in past (I have same module and same everything ) 

### Reproduction

1.attach module
2.switch flipper on 
3go gpio turn 5v on 
4.go apps and deauther 
5. app not working !

xMasterX, 2023-06-02:
Fixed in dev branch, thanks!

Also you don't need to go into GPIO and turn 5v manually, plugin enables 5v power automatically

DavidIgor, 2023-06-02:
@xMasterX 
aha ... 0.51 will fix it ?

xMasterX, 2023-06-02:
> @xMasterX aha ... 0.51 will fix it ?

yes