DarkFlippers/unleashed-firmware #499: TPMS modulation

DavidIgor, 2023-06-03:
### Describe the enhancement you're suggesting.

would be awesome to have modulation for TPMS car sensors, modulation is enough for me .... but if somebody wants to create app ...that would be something !

Crsarmv7l, 2023-06-10:
Protoview decodes a few. I had a Modulation for PMV-107J. Capturing is one thing, getting info from the captures is another. I made a python tool to generate subs for 4ish of the most common TPMS types. Tying to do that in C, on the flipper would be something... since they are 2-fsk, manchester encoded, with crc's.

DavidIgor, 2023-06-10:
@Crsarmv7l thanks, I managed to run your tool on mac and successfully get sub file for citroen tpms (not tested as I don't have a citroen). However my schrader tpms ID is not accepted by your tool for some reason

Crsarmv7l, 2023-06-10:
> @Crsarmv7l thanks, I managed to run your tool on mac and successfully get sub file for citroen tpms (not tested as I don't have a citroen). However my schrader tpms ID is not accepted by your tool for some reason

There are multiple different Schrader TPMS. I only implemented a single one, the SMD3MA4 if you are trying one that is not SMD3MA4 it wont work. Use rtl_433 to get the id.

DavidIgor, 2023-06-10:
@Crsarmv7l will try

DavidIgor, 2023-06-13:
@Crsarmv7l here is what hack rf spitted out : 20230605193820  433920000 FSK 38400 19200 Schrader D93437ABCFE0FFFF4B16/000000000000000000D6

I assume that "ABCFE0" is sensor ID?

S474N, 2026-02-02:
Wouldn't this help?
https://github.com/antirez/protoview