DarkFlippers/unleashed-firmware #503: ProtoView issue

DavidIgor, 2023-06-05:
### Describe the bug.

ProtoView after saving subghz file ....flipper cannot parse file 

### Reproduction

switch on 
open protoview
capture signal 
save the signal
go back to subghz and saved 
open saved file from protoview

ErSeraph, 2023-06-06:
Same issue, version 52

ErSeraph, 2023-06-06:
I leave here a potential fix I found for those who have the same problem, I haven't been able to try it yet. A comment that i wrote in the git page of the official ProtoView:
https://github.com/antirez/protoview/issues/14#issuecomment-1578910077

xMasterX, 2023-06-07:
Thanks for info, issue is present only with custom modulations, will push fix soon

closed 2023-06-07 by xMasterX

DavidIgor, 2023-06-12:
@xMasterX Hi , thanks for quick fix ... however it still needs to add end of the custom modulation (00 00 C0 00 00 00 00 00 00 00)in to saved file. 
- when created signal and replayed in app ... it does not taking created "Tire ID" it just spits out something like this: FC000000000000000000/03FFFFFFFFFFFFFFFFFF as tire ID  (captured with hackRF) 
- also when captured signal by app is replayed then it is not recognized by another flipper or car that signal was from.

xMasterX, 2023-06-13:
> @xMasterX Hi , thanks for quick fix ... however it still needs to add end of the custom modulation (00 00 C0 00 00 00 00 00 00 00)in to saved file.
>
> * when created signal and replayed in app ... it [...]

Hi, I made fix to add last part of modulation in files, currently in dev, sadly I didn't noticed it earlier so it will be in next release, or install dev build to test it right now

DavidIgor, 2023-06-13:
@xMasterX perfect! what about second part where protoview created tire ID has random ID?  I used hackrf portapack (tpms app) and recorded signal from protoview .... (after I manually fixed modulation)

xMasterX, 2023-06-13:
> @xMasterX perfect! what about second part where protoview created tire ID has random ID? I used hackrf portapack (tpms app) and recorded signal from protoview .... (after I manually fixed modulation)

Currently I don't know about random ID issue, ill look into it later

DavidIgor, 2023-06-14:
@xMasterX  installed latest FW 0.54  and protoview does saves the generated file with correct modulation , but the TPMS ID is just random no matter what you fill in while creating signal in protoview app. This is captured from Protoview and from saved sub file : (captured by tpms app in portapack h2)  20230613132932  433920000 FSK 38400 19200 Schrader 8A000000000000000000/89FFFFFFFFFFFFFFFFFF

note : 
-this is the same sensor (captured from sensor it self with tpms app in portapack H2) : 00000102065014  433920000 FSK 38400 19200 Schrader C13F3BABCFE0FFFF8314/000000000000000000D2

-above signal from tpms sensor is also picked up by protoview app and it decodes it correctly !  tpms id :ABCFE0