DarkFlippers/unleashed-firmware #508: Flipper freeze when emulating Mifare DESfire on a STID card reader

dreamshape, 2023-06-07:
### Describe the bug.

I work at a place where you access doors with your pass with STID readers and Simons Voss readers
we use Mifare DESfire 7b however the readers only read the UID that's not encrypted

I cloned/wrote my badge on a 7b keyfob n it work just fine however when i use the FZ to emulate the key it works on the Simons Voss readers without issues but on the STID readers it works but the screen get stuck on the emulation screen n i gotta reset it everytime after using it

issue doesn't happen when i use a magic nfc

Same issue both on latest UL and Xtrem fw

Another odd thing is maybe 1/4 times if i swipe quickly enough for the STID to read my FZ but not too long it doesn't get stuck so idk if it's something about the reader sending something /after/ recognizing the card that makes the FZ freeze

### Reproduction

1. go to nfc
2. emulation
3. swipe the FZ for more than 1,5 seconds
4. it works
5. it get stucks
6. i reboot the fz
7. repeat each time i need to use the elevator reader

### Target

STID NFC readers

xMasterX, 2023-06-07:
Can you verify it on latest OFFICIAL firmware? latest dev or release

dreamshape, 2023-06-08:
eh right nvm I checked with the official fw it does the same

xMasterX, 2023-06-08:
> eh right nvm I checked with the official fw it does the same

Can you please create same issue in official firmware and attach your reader photo or model?
Also your nfc file if possible

dreamshape, 2023-06-08:
sure I'll do that on the forum.flipperzero thing

UrSuLa360, 2023-06-10:
The bug happens with other readers, like a dl533n usb nfc reader with libnfc.
When the reader interacts with flipper emulating desfire uid, it gets frozen.
If you don't interact with any reader, bug doesn't happen.

dreamshape, 2023-06-10:
@UrSuLa360 by that you mean that it'll get unfrozen if i wait for a while without rebooting?

dreamshape, 2023-06-10:
there's the nfc file just wondering would removing everything under mifare desfire specific data changes anything ? since the reader only check the uid

<details>
<summary>Details</summary>

# Nfc device type can be UID, Mifare Ultralight, Mifare Classic, Bank card
Device type: Mifare DESFire
# UID, ATQA and SAK are common for all formats
UID: 04 5F 3E 0A 29 6F 80
ATQA: 03 44
SAK: 20
# Mifare DESFire specific data
PICC Version: 04 01 01 12 00 1A 05 04 01 01 02 01 1A 05 04 5F 3E 0A 29 6F 80 CF 0E 56 45 30 28 20
PICC Free Memory: 8192
PICC Change Key ID: 00
PICC Config Changeable: true
PICC Free Create Delete: true
PICC Free Directory List: true
PICC Key Changeable: true
PICC Max Keys: 01
PICC Key 0 Version: 00
Application Count: 0

</details>

UrSuLa360, 2023-06-10:
> @UrSuLa360 by that you mean that it'll get unfrozen if i wait for a while without rebooting?

no, once you interacted with the reader the flipper gets unresponsive.

I just wanted to point, flipper is responsive until any communication with the reader is performed

nahaki, 2023-06-14:
Thanks for sharing

UrSuLa360, 2023-09-08:
today i tested it with 62e and it worked correctly.
can you still reproduce this bug @dreamshape with latest version?

dreamshape, 2023-09-08:
It didn't work with a clone I previously made but I recloned it with the latest version and indeed it work so that's cool :vwonder what did they changed I might try to compare the old and new .NFC file to see what's different

closed 2023-09-08 by dreamshape