DarkFlippers/unleashed-firmware #515: NFC read fails to write a known Mifare Classic key to .nfc file in Unleashed-053

tsurmacz, 2023-06-12:
### Describe the bug.

Scenario: Start NFC main app, read the contents of the Mifare Classic card, save it. Key for sector 2 (blocks 8-11) is already in User Dict, key for sector 3 is unknown.

After upgrading from unleashed 0.51e to 0.53e the key B is now not written to a saved .nfc file.
After reverting back to 0.51e and reading the card again - the key is saved correctly.

Here is the diff of the same Card.nfc file written by version 0.51 and 0.53:

```
 Block 8: 13 00 00 00 EC FF FF FF 13 00 00 00 08 F7 08 F7                       
 Block 9: 31 30 30 30 30 31 34 35 34 34 33 30 30 32 30 31 
 Block 10: D5 96 20 47 12 55 90 11 6A C3 4B B9 00 18 20 01
-Block 11: ?? ?? ?? ?? ?? ?? 08 77 8F 00 A4 1D BE F0 20 D8
+Block 11: ?? ?? ?? ?? ?? ?? 08 77 8F 00 ?? ?? ?? ?? ?? ??
 Block 12: ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ??
```

Note the difference in Block 11/Key-B part. Blocks 8-10 have been read correctly, so the correct key has been used, but it was not written to the saved .nfc file

### Reproduction

Start NFC app, read the contents of the card, save it. 
Saved file does not contain Key-B used for reading the sector information.

### Anything else?

This may be related to the change described in the 053 release:
"OFW: Fix reading Mifare Classic cards with unusual access conditions and fix emulation of unknown keys"

tsurmacz, 2023-06-16:
Update:
unlshd-052 works fine, unlshd-054 suffers the same problem, so the bug was definitely introduced in the 053 version.

Also - I made some other tests, after recovering additional keys.  It looks, that the problem manifests itself when Access Bits are set to 08 77 8F, key-B is known but key-A is not known -- then key-B is not written to card dump (and also - shown immediately  as ?? with "Info" option in flipper).  If both keys are known, then both of them are written properly to the .nfc file.

xMasterX, 2023-06-18:
> Update: unlshd-052e works fine, unlshd-054e suffers the same problem, so the bug was definitely introduced in the 053e version.
> 
> Also - I made some other tests, after recovering additional keys. It looks, that the problem manifests itself when Access Bits are set to 08 77 8F, key-B is known but key-A is not known -- then key-B is not written to card dump (and also - shown immediately as ?? with "Info" option in flipper). If both keys are known, then both of them are written properly to the .nfc file.

Previous releases like 052 had wrong logic in this place, incorrect key validation was used
Can you verify if all works on latest official firmware dev build?

Leptopt1los, 2024-01-06:
@tsurmacz We are closing this issue due to its age and inactivity. please recreate the issue if the problem occurs on the latest release

closed 2024-01-06 by Leptopt1los