DarkFlippers/unleashed-firmware #517: furi_check failed

murph146, 2023-06-14:
### Describe the bug.

When opening ESP32-Wifi Marauder app closing and then re-opening you get the error "flipper crashed and was rebooted -  furi_checked failed." Tried on multiple Flipper Zeros running v053 and v054. Apologies if this issue is in the wrong place.  

Tested with Marauder firmware 10.6 and 10.7

### Reproduction

1. Power on flipper
2. Navigate to Applications > GPIO > ESP32-Wifi Marauder
3. Close app
4. Re-open
5. Crash Error

xMasterX, 2023-06-14:
Thanks, fixed in dev branch, will be in release 055

closed 2023-06-14 by xMasterX

murph146, 2023-06-14:
Much appreciated!!

Dominua, 2023-08-20:
Happening every second esp32 app open on 062r