DarkFlippers/unleashed-firmware #518: SubGHZ Frequency Analyzer captured signals list

sethrice, 2023-06-14:
### Description of the feature you're suggesting.

Curious if we can extend some existing functionality around the Frequency Analyzer. The official Frequency Analyzer had in it a single column list. 3 historical recorded frequencies, with the current one as the 4th on top. 

[image]

I discovered just before switching to the unleashed-firmware, that with the official firmware, you could scroll down in that single column list. It'd show you every frequency picked up as you ran it. With a count of how many times it saw each.

[image]

I really like your setup, with the exception that I can't scroll to see more historical found frequencies. I realize adjusting the RSSI can help with this, if I know what frequencies I'm looking for. However there are times when I get flooded with other frequencies with stronger signals, and that wipes out the count for the frequency i'm looking to identify. Also there are times when I don't know the frequency i'm looking for, due to hopping. So I'm just dropping the RSSI limit all the way left to see what's out there. That might be related to what I'm testing. I've attempted to make tweaks to get the scrolling option going with what you all have, based on the official version. But my coding skills are very weak in this area apparently. Thus the feature request to be done by a more skilled person(s).

xMasterX, 2023-06-14:
Frequency analyzer is not intended to use as "signal detector"

This is tool to get your remote frequency, it works only in around 5cm range, you need to put your remote very close to your flipper

signals you getting are mostly noise and other garbage, keeping it more than in 4 slots sounds as fully unneeded feature

sethrice, 2023-06-15:
That's a bummer, sounds like it's perhaps too difficult to shoehorn in, or
not worth the squeeze to restore that piece, no worries. I get it. I'm not
asking for a refactor by any stretch as that kind of thing is a PITA.

This feature is in the official firmware build that this repo is forked
from. (or was back in March 2023) So it has value or it wouldn't have been
included in the official build. I fully understand porting over
functionality isn't always as easy as one might hope. I really like the
overhaul you all did with it overall. Which is why I put in the request
instead of doing it myself with a PR.

FYI - saying broadly it's a "fully unneeded feature" though is like being
the guy on the tech board who's only contribution is "Why would you want to
do that, that's stupid", instead of trying to actually help the person find
an answer. You may not need it, or understand why someone might want to do
things a certain way. But others might have ideas you don't, where they can
utilize the feature differently to bring even more value to what you've all
built. Or they wouldn't be asking about it. ;)

How I'm using it, occasionally off-label shots in the dark, isn't really
relevant, outside of showing the value that restoring that feature presents
to my adhoc discovery workflow. Having helped beta SANS SEC556, I'm no
stranger to any of the more standard various tools, hardware/software, and
techniques for doing this with a laptop or even with a pi. I could do the
same, and then some with a hackrf, laptop, and URH. So yes, there are other
arguably better ways to accomplish the same thing when you are planning for
it. But that's inconvenient to carry around for those spontaneous moments
of inspired discovery. So nerfed internal flipper antenna or not; being
able to detect, then record and demodulate a specific frequency on the fly
is a huge reason why I love this tool. Plus it fits in my pocket. A
win-win...win!

I realize you all get a ton of requests and have to say no to a lot. There
may even be limitations I don't fully understand regarding the hardware
mem/proc being able to handle 50 slots with your additional existing
tweaks. Restoration of the 50 slots instead of just the current 4 seemed
reasonable enough to ask about at least. :shrug:

Appreciate your time, hope the insight helps for future feature development
consideration. Never hurts to have perspective for use cases you'd
otherwise never think to consider. Appreciate all you've done on this
project.