DarkFlippers/unleashed-firmware #521: Phone app compatability

SierraActual, 2023-06-16:
### Describe the bug.

It seems with the latest version, there are phone app issues. When I completely re-flash the unleashed OS, the phone app just gets stuck on "Syncing 0%" and I can no longer open files or remote control flipper until I re-flash official firmware. I have tried completely forgetting both the phone on the flipper and the flipper on the phone to reset the bluetooth connection and it doesnt help. For reference, I have uberguido's repository loaded on it as well.

### Reproduction

1. install latest official flipper firmware via phone app
2. sync all folders on iPhone app
3. forget flipper on iPhone app
4. forget flipper on bluetooth settings in phone
5. forget bluetooth devices on flipper
6. install unleashed 054 via web updater
7. open flipper app on iPhone (latest software update)
8. click connect flipper on iPhone
9. turn bluetooth on on flipper
10. click connect and run through connection steps on phone
11. gets stuck at Syncing 0%

### Logs

```Text
# Log when disconnecting and reconnecting to iPhone app (may not be initiating file sync at all):

266483 [I][BtKeyStorage] Base address: 200301E4. Start update address: 20030238. Size changed: 324
266485 [I][SavedStruct] Saving "/int/.bt.keys"
266489 [I][BtGap] Disconnect from client. Reason: 13
266495 [I][BtSrv] Close RPC connection
269820 [I][BtGap] Connection parameters: Connection Interval: 24 (30 ms), Slave Latency: 0, Supervision Timeout: 72
270055 [I][BtGap] Pairing complete
270060 [I][BtSrv] Open RPC connection
270235 [I][BtGap] Rx MTU size: 414
280999 [I][Loader] Starting Bluetooth
281004 [I][SavedStruct] Loading "/int/.bt.settings"
281009 [I][AnimationManager] Unload animation 'L3_FlipperMustache_128x64'
282149 [I][BtGap] Stop advertising
282154 [E][BtGap] set_non_discoverable failed 12
282203 [I][BtGap] Disconnect from client. Reason: 16
282207 [I][BtSrv] Close RPC connection
284593 [I][BtGap] Start advertising
284605 [I][BtGap] Connection parameters: Connection Interval: 24 (30 ms), Slave Latency: 0, Supervision Timeout: 72
284899 [I][BtGap] Pairing complete
284906 [I][BtSrv] Open RPC connection
284989 [I][BtGap] Rx MTU size: 414

# Stops here until new input
```

UrSuLa360, 2023-06-16:
i tried to connect a flipper with unleashed 054e to latest iphone app Flipper Mobile App 1.5.0 (in an iPhone SE 3rd gen,  iOS 15.7.1).
Everything worked fine, i installed the iOS app, enabled bluetooth in both devices and started the pairing. It worked and synced all my data ok.

@SierraActual  Can you add your iphone model and iOS version?

`[17:28:08][info]: app version: 1.5.0(243)
[17:28:08][info]: log level: debug
[17:28:08][info]: connecting
[17:28:08][info]: session started
[17:28:10][info]: connected
[17:28:10][info]: synchronize
[17:28:10][debug]: > request(storage(read(/int/.region_data)))
[17:28:10][debug]: > request(system(setDate(2023-06-16 15:28:10 +0000)))
[17:28:10][debug]: < response(storage(stat(49767)))
[17:28:10][error]: verify region: StorageError: File/Directory does not exist
[17:28:10][debug]: < response(storage(info(24 KiB / 44 KiB)))
[17:28:10][debug]: > request(storage(info(/ext)))
[17:28:10][debug]: < response(ok)
[17:28:10][info]: syncing date: (368s)
[17:28:10][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[17:28:10][debug]: < response(storage(info(216 MiB / 29.7 GiB)))
[17:28:10][debug]: > request(storage(list(/any)))
[17:28:10][debug]: < response(storage(list([mifare_fuzzer, mag, update, lfrfid, ibutton, Manifest, apps, badusb])))
[17:28:10][debug]: < response(storage(list([dolphin, infrared, apps_data, nfc, subghz, u2f, favorites.txt, badbt])))
[17:28:10][debug]: < response(storage(list([authenticator, subghz_remote, ibtnfuzzer, music_player, schip, wav_player, brainfuck, vb_migrate])))
[17:28:10][debug]: < response(storage(list([unitemp, spimem, rfidfuzzer, subplaylist, swd_scripts])))
[17:28:10][debug]: > request(storage(list(/any/subghz)))
[17:28:10][debug]: < response(storage(list([assets, tmp_history, XXXX.sub, XXXX.sub, XXXX.sub, XXXX.sub, XXXX.sub, XXXX.sub])))
[17:28:10][debug]: > request(storage(list(/any/lfrfid)))
[17:28:11][debug]: < response(storage(list([XXXX.rfid, XXXX.rfid])))
[17:28:11][debug]: > request(storage(list(/any/nfc)))
[17:28:11][debug]: < response(storage(list([assets, XXXX.nfc, debug.txt, XXXX.nfc, .cache, XXXX.nfc, XXXX.nfc, XXXX.nfc])))
[17:28:11][debug]: < response(storage(list([XXXX.nfc, .nested, XXXX.nfc, XXXX.nfc])))
[17:28:11][debug]: > request(storage(list(/any/nfc)))
[17:28:11][debug]: < response(storage(list([assets, XXXX.nfc, debug.txt, XXXX.nfc, .cache, XXXX.nfc, XXXX.nfc, XXXX.nfc])))
[17:28:11][debug]: < response(storage(list([XXXX.nfc, .nested, XXXX.nfc, XXXX.nfc])))
[17:28:11][debug]: > request(storage(list(/any/infrared)))
[17:28:11][debug]: < response(storage(list([assets, Consoles, Fans, Heaters, Miscellaneous, Projectors, Speakers, ACs])))
[17:28:11][debug]: < response(storage(list([TV_Tuner, TVs, Videoconferencing, ir_remote])))
[17:28:11][debug]: > request(storage(list(/any/ibutton)))
[17:28:11][debug]: < response(storage(list([])))
[17:28:11][debug]: > request(storage(hash(/any/subghz/XXXX.sub)))
[17:28:11][debug]: < response(storage(hash(1c48964deef36984ec204a2d43fe063b)))
[17:28:11][debug]: > request(storage(hash(/any/subghz/XXXX.sub)))
[17:28:12][debug]: < response(storage(hash(1f0bd0e9138500cfa834ae828e7596db)))
[17:28:12][debug]: > request(storage(hash(/any/subghz/XXXX.sub)))
[17:28:12][debug]: < response(storage(hash(0cf5506ada74a7e7a3eb7e569709fa25)))
[17:28:12][debug]: > request(storage(hash(/any/subghz/XXXX.sub)))
[17:28:12][debug]: < response(storage(hash(599a9e07c3146eb6f350dfbe652cdb01)))
[17:28:12][debug]: > request(storage(hash(/any/subghz/XXXX.sub)))
[17:28:12][debug]: < response(storage(hash(eb47e9d36bc61efc5dc4762fed9bd3d7)))
[17:28:12][debug]: > request(storage(hash(/any/subghz/XXXX.sub)))
[17:28:12][debug]: < response(storage(hash(ab77777da1f364288094b6b5b584d0e6)))
[17:28:12][debug]: > request(storage(hash(/any/lfrfid/XXXX.rfid)))
[17:28:12][debug]: < response(storage(hash(0cefb0677ac5ceec2d4bb8710170e14d)))
[17:28:12][debug]: > request(storage(hash(/any/lfrfid/XXXX.rfid)))
[17:28:12][debug]: < response(storage(hash(fe49c26f5d0ce67386f09e59c65547cc)))
[17:28:12][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[17:28:12][debug]: < response(storage(hash(b7d69bdb2ba325ec5b88a3111d0115c5)))
[17:28:12][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[17:28:12][debug]: < response(storage(hash(a0a6390a3865c4fbde17320d09152e31)))
[17:28:12][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[17:28:12][debug]: < response(storage(hash(dc082a735eae6c00d058b201fef52d79)))
[17:28:12][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[17:28:12][debug]: < response(storage(hash(ec4dff1a67533aa191f74d7d3c0807c5)))
[17:28:12][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[17:28:12][debug]: < response(storage(hash(60b13dc8b3693bb8fb343736de2d20c7)))
[17:28:12][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[17:28:13][debug]: < response(storage(hash(47e29b3d43704082d73094ea631470bf)))
[17:28:13][debug]: > request(storage(hash(/any/nfc/XXXX.nfc)))
[... 231 lines cut ...]
[17:41:11][debug]: < response(system(property(gauge.chginh: 0)))
[17:41:11][debug]: < response(system(property(gauge.fc: 0)))
[17:41:11][debug]: < response(system(property(gauge.otd: 0)))
[17:41:11][debug]: < response(system(property(gauge.otc: 0)))
[17:41:11][debug]: < response(system(property(gauge.sleep: 0)))
[17:41:11][debug]: < response(system(property(gauge.ocvfail: 0)))
[17:41:11][debug]: < response(system(property(gauge.ocvcomp: 1)))
[17:41:11][debug]: < response(system(property(gauge.fd: 0)))
[... cut ...]
[17:41:12][debug]: < response(system(property(gauge.capacity.design: 2101)))
[17:41:12][debug]: < response(system(property(gauge.capacity.remain: 1893)))
[17:41:12][debug]: < response(system(property(gauge.state.charge: 91)))
[17:41:12][debug]: < response(system(property(gauge.state.health: 100)))
[17:41:12][debug]: < response(system(property(gauge.voltage: 4100)))
[17:41:12][debug]: < response(system(property(gauge.current: 830)))
[17:41:12][debug]: < response(system(property(gauge.temperature: 31)))
[17:41:12][debug]: < response(ok)`

SierraActual, 2023-06-16:
iPhone 14 Pro - iOS 16.5

Are you running loads from the uberguido DB? I'm wondering if something in there is causing it to freeze up maybe. I may wipe my microSD and try to run the app without any loads, then slowly add sections if it works to maybe narrow down if that's the case.

SierraActual, 2023-06-16:
I pulled logs from the app. Something to do with my /any/infrared/ folder? 

‘’’ [12:19:52][info]: app version: 1.5.0(243)
[12:19:52][info]: log level: debug
[12:19:52][info]: connecting
[12:19:52][info]: session started
[12:19:53][info]: connected
[12:19:53][info]: synchronize
[12:19:53][debug]: > request(storage(read(/int/.region_data)))
[12:19:53][debug]: > request(storage(stat(/ext/Manifest)))
[12:19:53][debug]: > request(system(setDate(2023-06-16 16:19:53 +0000)))
[12:19:53][debug]: < response(storage(stat(49767)))
[12:19:53][error]: verify region: StorageError: File/Directory does not exist
[12:19:54][debug]: < response(storage(info(24 KiB / 44 KiB)))
[12:19:54][debug]: > request(storage(info(/ext)))
[12:19:54][debug]: < response(ok)
[12:19:54][info]: syncing date: (486s)
[12:19:54][debug]: > request(storage(stat(/ext/nfc/.mfkey32.log)))
[12:19:54][debug]: < response(storage(info(1.4 GiB / 29.7 GiB)))
[12:19:54][debug]: > request(storage(list(/any)))
[12:19:54][debug]: < response(storage(list([update, lfrfid, dolphin, Manifest, apps, apps_data, badusb, u2f])))
[12:19:54][debug]: < response(storage(list([ibutton, ibtnfuzzer, nfc, music_player, subghz, subplaylist, rfidfuzzer, System Volume Information])))
[12:19:54][debug]: < response(storage(list([favorites.txt, subghz_remote, Infrared, swd_scripts, .Trash-1000, README.md])))
[12:19:54][debug]: > request(storage(create(/any/infrared, true)))
[12:19:54][critical]: synchronize: StorageError: File/Directory already exists
[12:19:54][error]: synchronize: StorageError: File/Directory already exists’’’

SierraActual, 2023-06-16:
Update: Reformatted SD card, re-flashed Unleashed, reloaded loads from DB. All working fine in app now. Thanks!!

closed 2023-06-16 by SierraActual