DarkFlippers/unleashed-firmware #532: User Dict MF Keys multiple deleted

cozzm0, 2023-07-04:
### Describe the bug.

When deleting one key, two keys are deleted at a time instead of just the one key. 

### Reproduction

Open Mifare Classic Keys
View list (user dict)
Select first key on list
Press right to delete and first key and second key will be deleted. 

### Anything else?

Have had 3 keys deleted at once (list went from 5 to 2 keys when playing around, but I don’t recall how o made that happen). 
Version 052

UrSuLa360, 2023-07-04:
Yes, reproducible in 055 too.

Deleting a key also remove some extra "\n" or "\r\n" 

From: (5 keys)

key1
key2
key3
key4
key5

To: (3 "keys")

key1
key2key4
key5

Nice catch @cozzm0

xMasterX, 2023-07-05:
Fixed in dev, thanks

closed 2023-07-05 by xMasterX