DarkFlippers/unleashed-firmware #542: Broken infrared recording

dwroland, 2023-07-12:
### Describe the bug.

After upgrading from 52e to the latest 56e, the flipper cannot record IR correctly. It does not recognize the NECext protocol, and the recorded RAW is not recognized. Emulation of signals recorded on the 52e version works correctly.

### Reproduction

1. Go to Infrared > Learn New Remote
2. Record signal (it will not recognize NECext, not sure about other protocols)
3. Send recorded RAW signal

dwroland, 2023-07-22:
Flipper is able to read NECext correctly once every around hundred times. It reads other remotes such as those from the TV without any problems.

xMasterX, 2023-08-17:
NECext reads correctly on 060 release, can you provide more details?
I made raw recordings of the necext remote and sent it from one flipper to another, it gets recognised correctly
Not as raw

dwroland, 2023-12-10:
It took me a while.
The problem is not related to the release version. I checked various forks and versions. It was just a coincidence that it worked for a while. I still don't understand why it doesn't work. 
I took a moment to analyse the raw data using this https://techdocs.altium.com/display/FPGA/NEC+Infrared+Transmission+Protocol and it seems fine.
Changing the frequency value in remote.ir changes nothing. Unfortunately I do not have a second flipper to analyse the difference between the outputs of the raw and parsed records.

[image]

dwroland, 2023-12-17:
[image]