DarkFlippers/unleashed-firmware #547: Authenticator app error + flipper crash

SilentMonster0, 2023-07-14:
### Describe the bug.

from what i know, on version 056e, when you open the authenticator app you get an error that says "an error has occured during loading config file" and when you go back, the flipper crashes with a "null pointer dereference"

[image]
[image]

### Reproduction

1. go to applications
2. go to tools
3. open the authenticator app
4. you get an error and flipper crashes then you go back

UrSuLa360, 2023-07-15:
it is working for me, i had no previous tokens. I can enter the app and create new tokens. Later i can exit/enter the app multiple times and check the auth code  without problem.

Did you have previous tokens saved before upgrading to the lastest version?
What was the previous version you was using before upgrading to 056e?

SilentMonster0, 2023-07-16:
i dont think i have any tokens. 
the previous version I was using before 056e was the one before it (055e)

SilentMonster0, 2023-07-16:
**UPDATE: i fixed the bug lol all i had to do was delete the authenticator folder and also deleted the apps folder just to be safe, I can use it now**

closed 2023-07-16 by SilentMonster0