DarkFlippers/unleashed-firmware #548: Firmware corrupts Internal Flash but no negative effect except no file transfer via app

QuestionQuest117, 2023-07-15:
### Describe the bug.

After installing Flipper Unleashed, in the mobile app, the internal flash storage size indicator turns red and I don't have functionality to send files to my Flipper Zero via Bluetooth. Everything else still works and I can update it via my phone, but yea.

### Reproduction

• Install Flipper Unleashed
• Look at Android mobile app's Internal Flash display
• Go to options in the app
• Note that "File Transfer" is not there

### Anything else?

I made a post about this issue on the subreddit and others informed me that they were not encountering this bug. My app is up to date and Flipper is running the latest version of Unleashed. Tried reseting the flash and reinstalling Unleashed, but that didn't fix it.

GIGObug, 2023-07-16:
I’m having a similar issue with the app not connecting to flipper after the 056e update. I’m using iPhone 12 Pro Max and have also tried the app on my iPad Pro.

liszkai, 2023-07-16:
I had this issue too.
The solution was to backup 3 files (.desktop.settings; .dolphin.state; .notification.settings) from internal flash, then reset to factory defaults, then copy back these files. You will keep your Flipper level, just neet to repair the phone.

GIGObug, 2023-07-16:
Hi Liszkai,
I don't have a windows 10 PC. Instead I downgraded to the previous version
(055e), because it was downloaded to the MicrSD card. My flipper is
connected to the App again. Thanks for your help.

xMasterX, 2023-07-17:
Red color of free/used memory text doesn't mean its corrupted, that means its too low of free space
This is already fixed in dev branch and will be in next release, you should try dev build or wait for next release

closed 2023-07-17 by xMasterX