DarkFlippers/unleashed-firmware #555: Android app doesn't allow to start Emulate (NFC) or Send (SubGHz)

improving-rigmarole, 2023-07-24:
### Describe the bug.

Sorry, don't really know if this problem is known or if it's an isolated bug on my side, and/or if it's normal (I think that was what [this issue](https://github.com/DarkFlippers/unleashed-firmware/issues/127) tried to ask help for)
It works for official FW on my side 

App seemingly doesn't want to enable the button to allow to Emulate or Send when using non official firmware and asks to update

[image]

### Reproduction

Using
- Android app Version `1.6.1.1114` (on android 13) tried both beta and release apps from GooglePlay
- Flipper firmware `unlshd-058` (didn't work with previous ones either on my side)

1. Connect flipper via bluetooth
2. go to archive
3. select a SubGhz or NFC thing
4. button to Emulate/Send is disabled

xMasterX, 2023-07-25:
Issue is not present with iOS app, please report it directly to android app devs in their repo or using bug report form
This issue is not related to firmware, thats related only to current android app releases
Maybe wrong check for firmware version or something else is broken in the android app

improving-rigmarole, 2023-07-26:
ok so I just checked again, and the "latest version" mentioned wasn't Flipper's version but the app's version apparently,
since somehow the beta version isn't considered as latest...
fix was to get the public version from PlayStore and not the beta program one...

sorry for noise and thanks for the cfw !