DarkFlippers/unleashed-firmware #560: Write to populated MIFARE CLASSIC 1K card

rapbando, 2023-07-27:
### Describe the enhancement you're suggesting.

Right now is only possible to write a properly saved nfc card to a initlal card (bank or virgin), while other [tools](https://play.google.com/store/apps/details?id=de.syss.MifareClassicTool)  are able to overwrite the same card and blank ones.

wosk, 2023-07-28:
Use "Nfc magic" application to write gen1a/b and gen4 mifare tags

closed 2023-07-28 by xMasterX

rapbando, 2023-07-29:
Under NFC magic there is only two option:

- Check magic tag
- Authenticate Gen4

No option to write, let alone a MIFARE Classic 1K...

liszkai, 2023-07-29:
The next menu depend on your magic tag version. Use "Check magic tag", then you can write it.
[image]
[image]
[image]

rapbando, 2023-07-29:
@liszkai 

That only works for magic tags not for MIFARE Classic 1K...

@xMasterX  Please reopen the issue because is not solved.