DarkFlippers/unleashed-firmware #563: New update blocking the backlight from going on

TheDarkOn3, 2023-07-28:
This issue is noticed with the 59r.tng file. Was the code accidentally removed from the new update?

xMasterX, 2023-07-28:
User related issues like this one, can not be fixed, I don't know who could fix inability to read
Do you understand that other color LEDs will not grow up magically in your flipper? 
How many times I need to add this (only for hardware modded flippers) so it won't be ignored again and again?

closed 2023-07-28 by xMasterX

TheDarkOn3, 2023-07-29:
I am saying it’s no lighting up whatsoever but still have the image. It’s something in the code that causes this. I am not saying anything about the color mod. Sent from my iPhone

SkeletonMan03, 2023-07-29:
You didn't read the instructions. The version that has r at the end is for Flipper Zeros that have had their screens modded with an RGB board. You do not have that RGB mod. Therefore, you get no backlight with the r version installed. Install one of the other versions. Actually read what it says above the files on the release page.

TheDarkOn3, 2023-07-29:
I noticed that myself (I’m kinda noob with some knowledge on programming). I just saw that. To everyone thx for the help. Sent from my iPhone