DarkFlippers/unleashed-firmware #567: [NFC] Fail to fully read a mifare classic 1k card with valid keys in user dictionary

UrSuLa360, 2023-08-04:
### Describe the bug.

I read Mifare classic card 1k with 5 unknown B keys. 
[image]

Now i try to recover keys with Mifare Nested by @AloneLiberty 
[image]

`python -m FlipperNested
[?] Checking 0EFB2C68.nonces
[!] No nonces found
[!] Failed to parse 0EFB2C68.nonces
[?] Checking DCC57129.nonces
Recovering key type B, sector 11
Found 1 key(s): ['8ae675fb1067']
Recovering key type B, sector 12
Found 1 key(s): ['aa3a9cf1c6f7']
Recovering key type B, sector 13
Found 1 key(s): ['2595b1d36a23']
Recovering key type B, sector 14
Found 1 key(s): ['91f69383da4d']
Recovering key type B, sector 15
Found 1 key(s): ['0abf7fedf635']
[+] Found potential 5 keys, use "Check found keys" in app`

[image]

The mf_classic_dict_user.nfc after "check found keys" process results in: (the key  669957a21cb8 was already in dict)

669957a21cb8
8AE675FB1067
AA3A9CF1C6F7
2595B1D36A23
91F69383DA4D
0ABF7FEDF635

Then with keys in dict i read the card again.

[image]

Same result even with keys in dict.

[dumps.zip](https://github.com/DarkFlippers/unleashed-firmware/files/12260178/dumps.zip)

The keys are valid, i can unlock the card with moc in the computer. I attached both dumps, partial dump created by flipper and fully unlocked dump with moc in the computer.

Anything let me know. Thank you

### Reproduction

.

### Target

NFC

AloneLiberty, 2023-08-04:
As you pinged me, I see its B keys, might be fixed with https://github.com/flipperdevices/flipperzero-firmware/pull/2912. Just to be sure, can you check on latest dev OFW version?

UrSuLa360, 2023-08-04:
[image]

Latest dev OFW Worked.

Thanks! You're awesome!

(i think your changes will be released here soon, then i will recheck again and close the issue)

xMasterX, 2023-08-05:
> [image]
> 
> Latest dev OFW Worked.
> 
> Thanks! You're awesome!
> 
> (i think your changes will be released here soon, then i will recheck again and close the issue)

Can you check with latest dev build?
[Link to web updater](https://lab.flipper.net/?url=https://unleashedflip.com/fw/dev/flipper-z-f7-update-755.tgz&channel=dev-cfw&version=755)

UrSuLa360, 2023-08-05:
[image]

Latest unleashed dev version Worked! 

I got the version from this link
[https://lab.flipper.net/?url=https://unleashedflip.com/fw_extra_apps/flipper-z-f7-update-757e.tgz&channel=dev-cfw&version=757e](url)

PD: your link didn't worked, "Failed to fetch"

xMasterX, 2023-08-05:
> [image]
> 
> Latest unleashed dev version Worked!
> 
> I got the version from this link [https://lab.flipper.net/?url=https://unleashedflip.com/fw_extra_apps/flipper-z-f7-update-757e.tgz&channel=dev-cfw&version=757e](url)
> 
> PD: your link didn't work, "Failed to fetch"

yea build was a bit outdated
thanks!
all will be in 060 release, for now i'm closing this issue

closed 2023-08-05 by xMasterX