DarkFlippers/unleashed-firmware #582: Evil Portal on ESP Marauder App

malebuffy, 2023-08-20:
### Describe the bug.

When using the Unleashed Firmware with the built in ESP Marauder app, the built in Evil Portal app doesn't work. I have put the evil_portal folder on the apps_data folder as in the Evil Portal instructions. It finds the index.html file but although the ap_config.txt file is in the same folder, it dosn't recognise/finds it. Any pointers? 

Noter: Not talking about the standalone ESP32 Evil portal (which for some reason crashed for me)

### Reproduction

1. Select Apps
2. ESP32 Marauder
3. Evil portal
4. Error appears.

xMasterX, 2023-08-20:
What error appears when you select evil portal in marauder? and why do you think it should use same files and folders as regular evil portal app? what crashes do you get and steps to reproduce them?

malebuffy, 2023-08-21:
What error appears when you select evil portal in marauder?

Ap.config.txt not found

and why do you think it should use same files and folders as regular evil portal app? 

Because when I Recreate the evil_portal folder it stops lookikg for the index.html file and only searches for the ap.config.txt file. When the folder is missing the errror medsage is: cannot find index.html

what crashes do you get and steps to reproduce them?

With this version I do not get crashes. I get crashes if I install the standalone evil portal from flipc.org

massif-py, 2023-09-24:
`ssid -a -n uwu`

gphoenix14, 2023-09-25:
What is the folder where we ought to place index.html?

malebuffy, 2023-09-25:
> What is the folder where we ought to place index.html?

 apps_data/evil_portal

Banditwon, 2023-10-20:
Did you figure it out?

malebuffy, 2023-10-20:
unfortonately not. installed it standalone.

mwooller, 2023-10-23:
You don't need the ap.config.txt - select the ssid in the marauder app, and when you start evil_portal it will use that ssid

ruehlinger, 2023-11-15:
setting the ssid fixed the ap.config.txt issue I was having but now it is saying it cannot find index.html even though it is in apps_data/evil_portal. I even tried to move index.html to the html folder but to no avail. Any suggestions?

datadotlog, 2023-11-22:
In my case, it's not showing any errors, but don't go ahead of "Setting HTML"

dfucci, 2023-11-29:
> In my case, it's not showing any errors, but don't go ahead of "Setting HTML"

Same identical issue for me. Did you figure this out?

boobaloop, 2024-01-04:
@dfucci did you get this to work?

akashrpatil, 2024-01-15:
> setting the ssid fixed the ap.config.txt issue I was having but now it is saying it cannot find index.html even though it is in apps_data/evil_portal. I even tried to move index.html to the html folder but to no avail. Any suggestions?

Same issue

jstgtngstrtd, 2024-01-21:
Was this ever resolve?  I have found a workaround.

akashrpatil, 2024-01-22:
> Was this ever resolved? I have found a workaround.

Can you please help me on the same?

boobaloop, 2024-01-22:
I was able to get this to work by switching from a 64 gb microSDXC sd to a 16 gb microSDHC sd card

jstgtngstrtd, 2024-01-22:
Sure, I can try to help.  I am running the latest Unleashed Firmware.

-I placed the evil_portal file in SD Card/apps_data/marauder
-I cannot get the ap.config.txt to work at this point and I have not really tried that hard.  I will work on that at a later point.

Here is the workaround:
-Go to Apps -> GPIO ->[ESP32] Wifi Marauder-> SSID -> Select "add name" (to add an SSID for example "free google wifi") and save 

-Next go to List -> Select "ssid" and find number associated with the ssid you created ->Select change to "ssid" and enter the number associated with the ssid this number should be "0" at this point and save.

-Now go to "Load Evil Portal HTML file" and choose the file you would like to use.

-Final step, Go to Evil Portal and start.  At this point the IP address of your portal should be displayed.  Connect a device to the free google wifi and enter Username and Password and it should display on your Flipper.

Please note: Do not try to run Marauder at the same time as Evil Portal, this will not work.  I suggest clearing all lists before doing the above.  I hope this helps

akashrpatil, 2024-01-23:
> Sure, I can try to help. I am running the latest Unleashed Firmware.
> 
> -I placed the evil_portal file in SD Card/apps_data/marauder -I cannot get the ap.config.txt to work at this point and I have not really tried that hard. I will work on that at a later point.
> 
> Here is the workaround: -Go to Apps -> GPIO ->[ESP32] Wifi Marauder-> SSID -> Select "add name" (to add an SSID for example "free google wifi") and save
> 
> -Next go to List -> Select "ssid" and find number associated with the ssid you created ->Go to Select and change to "ssid" and enter the number associated with the ssid this number should be "0" at this point and save.
> 
> -Now go to "Load Evil Portal HTML file" and choose the file you would like to use.
> 
> -Final step, Go to Evil Portal and start. At this point the IP address of your portal should be displayed. Connect a device to the free google wifi and enter Username and Password and it should display on your Flipper.
> 
> Please note: Do not try to run Marauder at the same time as Evil Portal, this will not work. I suggest clearing all lists before doing the above. I hope this helps

Hey, Thank you for your detailed response. but still, I followed your steps twice but got stuck on that Setting HTML line :(. If it is possible can you please make a quick video and post it on YouTube that can be helpful for most of the people who are facing this issue Thank you so much for your help.

jstgtngstrtd, 2024-01-23:
> Hey, Thank you for your detailed response. but still, I followed your steps twice but got stuck on that Setting HTML line :(. If it is possible can you please make a quick video and post it on [...]

Hi there, Before I go and do that I would like to ask if you placed the evil_portal file in the correct folder.  You originally stated that you placed the evil_portal file in SD Card/apps_data, it needs to be placed in SD Card/apps_data/marauder.  Is this where you have the evil_portal file?

akashrpatil, 2024-01-23:
> Hi there, Before I go and do that I would like to ask if you placed the evil_portal file in the correct folder. You originally stated that you placed the evil_portal file in SD Card/apps_data, it [...]

Yes, I have placed Evil Portal file on the same path that is provided by you.

jstgtngstrtd, 2024-01-23:
> Yes, I have placed Evil Portal file on the same path that is provided by you.

Are the html files showing when you open the evil_portal file?

[image]

akashrpatil, 2024-01-23:
Hi Mate, While opening the evil portal I got this 
[image]
[image]

Even I try to set the HTML manually 
[image]

jstgtngstrtd, 2024-01-23:
I would take a few steps back.  Set your folder structure up like this.
[image]
[image]

akashrpatil, 2024-02-03:
> I wrote a how to on the steps that I took to get it to work. It seems like a lot but it only takes me about a minute to set up my own ssid and get this up and running.
> 
> [Evil portal on ESP Marauder.docx](https://github.com/DarkFlippers/unleashed-firmware/files/14033467/Evil.portal.on.ESP.Marauder.docx)

Hey Bro So sorry for the late revert. I have tried the same things which are mentioned in the document. On the Load Evil Portal HTML File I'm getting an error I guess there are some issues in the firmware itself😅
[image]

Wilfredo215, 2024-02-12:
Yeah i get the same thing @akashrpatil

Wilfredo215, 2024-02-12:
I found the issue. If you previously flashed it with the esp flasher then it will not work you have to flash it with the marauder flasher from https://github.com/UberGuidoZ/Flipper/blob/main/Wifi_DevBoard/FZ_Marauder_Flasher/ReadMe.md the the doc provided by @jstgtngstrtd works like a charm

akashrpatil, 2024-02-12:
Hi, @jstgtngstrtd @Wilfredo215 Thank you so much guys only because of your help, I'm able to set up the portal 😭

Eq1ofof, 2024-08-20:
After flashing the devboard like mentioned by @Wilfredo215  the problem disappeared and the issue can be closed.

dcohen-hub, 2024-12-06:
When I add SSID with <add Name> everything works. But I would like the added SSID's to be saved, so that I can just use List and seem them all the next time I power on flipper.
If I add 3 SSID's I can then List then Select until I power off/on.
Is there a file created on SD card when adding SSID's?

avipars, 2024-12-25:
> After flashing the devboard like mentioned by @Wilfredo215 the problem disappeared and the issue can be closed.

Where does ap.config go if I wanted to set the name that way rather than via the flipper keyboard?