DarkFlippers/unleashed-firmware #584: Firmware download modification dates wrong

mmiller7, 2023-08-20:
### Describe the bug.

It appears the downloads have a wrong file modification date, they report a build date of December 31, 1969, 19:00.  This makes it very hard to find the new files after extracting since a sort by modification date puts it before older firmware versions.

[image]

### Reproduction

Download the firmware tgz from https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-062
Open the archive to load into the SD card
Observe the file modification dates

pdostal, 2023-08-21:
@mmiller7 that's quite old firmware, indeed!

UrSuLa360, 2023-08-21:
Unix epoch - 5

That is not a bug, that file/folder has no date info. So your program/explorer just shows a default date the Unix epoch date with your current timezone (UTC-5). 
I guess you are from some place in America, right? :D

You can read some info about this [here](https://en.wikipedia.org/wiki/Unix_time)

mmiller7, 2023-08-22:
So why is that now being removed in the build process?

Filesystems don't just nothave a modified date...and I may not keep up with every release but the last 3 or 4 downloads of Unleashed I have installed did have proper data for the datestamps in the release package so clearly something is changed...and this serves no purpose but to increase confusion and be hostile to users by not having accurate data on the builds

FeyXieXzf, 2023-09-24:
> So why is that now being removed in the build process?
>
> Filesystems don't just nothave a modified date...and I may not keep up with every release but the last 3 or 4 downloads of Unleashed I have [...]

You can just look at the modified date & time of the firmware file *itself*.