DarkFlippers/unleashed-firmware #591: Impossible to turn off projectors using universal remotes

sashaMSU, 2023-08-28:
### Describe the bug.

The problem is that a lot of projectors require to press Power button twice in a short period of time (5-10 seconds) to be turned off. It is impossible to do so using universal remote with normal projectors.ir list because there each signal is listed only once. Restarting bruteforce will just cause projector timeout if the correct signal is not at the very beginning of the list. I tried duplicating every Power signal in projectors.ir and it solved the problem. I would be glad if you included list with doubled Power signals into firmware.

### Reproduction

1. Point flipper at a working projector
2. Go to universal remotes->projectors
3. Press Power button
4. At about 60% progress projector shows a message asking to press Power again
5. If I let flipper to continue, there is no more fitting signals
6. If I restart the process, it takes too long to reach the correct signal
7. After 10 seconds the message disappears and the projector stays on

### Anything else?

Here
[projectors.zip](https://github.com/DarkFlippers/unleashed-firmware/files/12458712/projectors.zip)
 is my list with doubled Power signals and a program that I wrote to make it.

mswietlicki, 2023-09-06:
Notice the same issue with Optoma HD29HST.

xMasterX, 2023-09-22:
Fixed in dev branch, thanks for info

closed 2023-09-22 by xMasterX