DarkFlippers/unleashed-firmware #593: Adding custom protocols

nickyg543, 2023-08-30:
### Describe the enhancement you're suggesting.

I'd like to define a SubGHz protocol for the flipper, but I'm not sure where to get the hex code for the CC1101 (like shown below)
[image]
Is there a guide anywhere describing how to do this? Use TI's SMARTRF studio?

nickyg543, 2023-08-31:
Update: adding to setting_user does not update the frequency list in the GUI, nor does it add new protocols
[image]
[image]
How should I go about adding protocols and frequencies?

wosk, 2023-09-03:
1. RTFM cc1101 datasheet for registry sequence coding (hex code) to add new protocol.
2. You have to uncomment Frequency line in setting_user file to enable it

xMasterX, 2023-09-05:
Please read @wosk response, and follow his recommendations

closed 2023-09-05 by xMasterX