DarkFlippers/unleashed-firmware #596: Add SubGHz Beninca's 128bit ARC support

kichro, 2023-09-05:
### Description of the feature you're suggesting.

Hey guys, I've found out that the "KL: Beninca 433MHz" flipper profile doesn't really work on any recent Beninca gates since they are moving from the legacy 64bit HCS rolling code to 128bit ARC rolling code. Any chance we could get this new profile on the Flipper? 

Official documentation available on this [link](https://www.beninca.com/download/en-guida-to-go-happy-iri-tx.html).

### Anything else?

I'm in possession of Beninca TOGO2VA remote and I'm willing to help with implementation of the profile. Let me know if you need samples or PCB photos.

[image]

kichro, 2023-09-14:
I'm closing this issue as I've figured out that reversing this proprietary protocol is probably well beyond the scope of this repository.

xMasterX, 2023-11-23:
> I'm closing this issue as I've figured out that reversing this proprietary protocol is probably well beyond the scope of this repository.

Yea, I'm not a magician who can do anything without any information and without hardware, signal samples or photos are not enough for 128bit rolling code protocol
Let's close this issue then

closed 2023-11-23 by xMasterX

martineliascz, 2023-12-22:
@xMasterX - Will the 128bit ARC support be ever added? Here is the [reddit post](https://www.reddit.com/r/flipperzero/comments/xi54n5/comment/keh5ndi/) I contributed to..

Thanks much!

bymatej, 2025-02-16:
What can we, mere mortals who don't understand much on the matter, do to contribute to adding 128-bit ARC support to Flipper? A few years back I opened [a thread](https://forum.flipper.net/t/beninca-to-go-2va-remote/17064) where I posted [some `.sub` files](https://github.com/bymatej/flipper-zero-beninca-sub/tree/main/sub_files), but I doubt they can help. Does anyone own Beninca remotes configured to use the new 128-bit that also owns a proper RTL device to capture some signal samples?

xMasterX, 2026-01-21:
Support added in latest dev commits, next dev builds after 1229 will have it built in
Please test and let me know how does it work for you, especially please try adding new remote via add manually
And binding that to receiver 

[image]

And one more thing, if someone has 4 buttons remote, please record RAW with every button in it
Also make second RAW with hidden button recorded
and share them here

[image]

After that it will be fully finished

bymatej, 2026-01-28:
I can't confirm or deny that the hidden button works. 
1. On my garage door, I don't need this, plus I don't want to experiment with it since I only have one remote
2. On my parking gate, the receiver is not the same as in my garage. I tried using the remote learn feature, but it did not work. It silently fails, and this could be due to remote learn being disabled on gates in buildings (for security reasons, I guess), or maybe it is just not supported. I think this is Beninca LADY gate.

But the remote can be paired to the receiver using the PGM button (at least to my receiver; there are many Bennica receivers out there). 

I know this wasn't helpful, but I just wanted to let you know that I tried and that I can't confirm the hidden button behavior.