DarkFlippers/unleashed-firmware #597: BLE ADV Packets

OEMatGIT, 2023-09-07:
### Description of the feature you're suggesting.

Hi,

is it possible to include BLE ADV Packets to spoof devices like AirTags or others like described here ?

https://techryptic.github.io/2023/09/01/Annoying-Apple-Fans/

i have compiled my own fw with the updated gap.c file which works fine and i would love to have that also in the unleashed fw

oliva, 2023-09-07:
Here's RogueMaster's implementation that makes it modular:
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/70b43369f0195fc5a86c7abe3652911792ba0627

xMasterX, 2023-09-07:
> Here's RogueMaster's implementation that makes it modular: [RogueMaster/flipperzero-firmware-wPlugins@70b4336](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/commit/70b43369f0195fc5a86c7abe3652911792ba0627)

First of all, RogueMaster has not made a single line of code in that app
App is being developed by Willy-JL
And App is already integrated in dev builds

closed 2023-09-07 by xMasterX