DarkFlippers/unleashed-firmware #598: Integrate Apple Bluetooth Low Energy (BLE) and ADV Packets Prank

zelorun, 2023-09-07:
### Description of the feature you're suggesting.

Apple devices can be spammed using bluetooth, see:

https://techryptic.github.io/2023/09/01/Annoying-Apple-Fans/

syxanash, 2023-09-07:
This would be a nice to have 😄

m1ga, 2023-09-07:
it's already included in the dev branch + extra apps.

and the same issue was posted right below yours https://github.com/DarkFlippers/unleashed-firmware/issues/597 @zelorun

zelorun, 2023-09-07:
@m1ga which command to use if i want to build locally with extra apps?

m1ga, 2023-09-07:
Check the `#unleashed-development` channel in discord. It has links to the webinstaller for every commit. Otherwise it is a normal build (commands are in the readme) and use the zip from https://github.com/xMasterX/all-the-plugins. Again: readme

closed 2023-09-07 by xMasterX