DarkFlippers/unleashed-firmware #601: Adding iPhone Bluetooth Device Popups

slashy, 2023-09-11:
### Description of the feature you're suggesting.

Would it be possible to add this as an select option or even as an app?

https://techryptic.github.io/2023/09/01/Annoying-Apple-Fans/

I think this would make for a nice addition

MiguelException, 2023-09-11:
Hi,

Take a look at #597 and #598

m1ga, 2023-09-11:
please use the search. It's already implemented

https://github.com/DarkFlippers/unleashed-firmware/issues/598
https://github.com/DarkFlippers/unleashed-firmware/issues/597

slashy, 2023-09-11:
> please use the search. It's already implemented
> 
> #598 #597

Oh dammit, I only searched in open as I didn't think it would be implemented so fast. Sorry! Closing down here...

closed 2023-09-11 by slashy