DarkFlippers/unleashed-firmware #604: Enhance BLE Attacks: Prompt for Bluetooth Activation When Disabled

Scrut1ny, 2023-09-12:
### Describe the enhancement you're suggesting.

If you disable Bluetooth in the settings and subsequently attempt to run BLE Attacks, they still execute despite Bluetooth being turned off. It would be more helpful if a prompt appeared, suggesting that you enable Bluetooth since it's currently switched off. Users might forget that they disabled Bluetooth and proceed to run the BLE Attack app, mistakenly thinking it's functioning when it's not.

nocerainfosec, 2023-09-13:
Apple BLE Spam seems not to be working, is it an issue with my flipper?

Scrut1ny, 2023-09-13:
> Apple BLE Spam seems not to be working, is it an issue with my flipper?

I don't think so, make sure Bluetooth is enabled and your on v63 of the firmware that just dropped. Try also restarting your flipper or flashing it again.

xMasterX, 2023-09-19:
Bluetooth is not required to be enabled, app enables and disables it by itself

closed 2023-09-19 by xMasterX