DarkFlippers/unleashed-firmware #625: Airtag spam

P6g9YHK6, 2023-10-11:
### Description of the feature you're suggesting.

Like the BLE apple spam it should just be a toggle that look for nearby airtag/tile/chip/etc and trigger the ring function every couple of seconds

i can do it from https://github.com/seemoo-lab/AirGuard so i suppose it should not be too complicated to implement an app for the flipper to do the same.

abjugard, 2023-11-02:
This seems actually useful, even if obviously super annoying in the hands of an immature manchild like myself.

AidenPearce69, 2023-12-07:
> This seems actually useful, even if obviously super annoying in the hands of an immature manchild like myself.

I second that motion and I second the immature manchild part as well I am known as a prankster in my family. Now with the Flipper Zero I can up my pranks. (INSERT DR. EVIL LAUGHTER HERE) :^)