DarkFlippers/unleashed-firmware #626: Updating to 64E Crashes USB connection

LuizizM, 2023-10-11:
### Describe the bug.

During the update both by USB of BT crashes the flipper giving it a time out only happens with the 64E package because the previous ones work fine.

### Reproduction

Connect flipper turned on on pc or bluetooth
selecte the 64e package and click on install
after a while it crashs with communication error

### Target

64E package

### Logs

```Text
707 [APP] qFlipper version 1.3.2 commit 7b0839e1 2023-06-08T07:58:22
708 [APP] OS info: Windows 11 Version 22H2 11 10.0.22621 Qt 6.4.2
723 [DBG] Trying serial port  at \\.\COM1
723 [DBG] Trying serial port FLIP_H0RU at \\.\COM3
723 [DBG] Using  serial port FLIP_H0RU at \\.\COM3
723 [RPC] Starting RPC session...
815 [RPC] RPC session started successfully.
816 [RPC] (1) System Protobuf Version START
873 [DBG] Detected protobuf version: 0.20
873 [RPC] (1) System Protobuf Version SUCCESS
873 [RPC] (2) Property Get START
996 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1018 [RPC] (2) Property Get SUCCESS
1018 [RPC] (3) Storage Info @/ext START
1122 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
4091 [RPC] (3) Storage Info @/ext SUCCESS
4091 [RPC] (4) Storage Stat @/ext/Manifest START
4154 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
4154 [RPC] (5) System Get DateTime START
4207 [DBG] Flipper time skew is -854 milliseconds
4207 [RPC] (5) System Get DateTime SUCCESS
4207 [RPC] (6) System Set DateTime START
4259 [RPC] (6) System Set DateTime SUCCESS
4263 [RPC] Stopping RPC session...
4468 [RPC] Attempting to unload protobuf plugin...
4468 [RPC] Unloaded protobuf plugin.
4468 [RPC] RPC session stopped successfully.
4468 [REG] Registering the device
4468 [DEV] Version: unlshd-063e commit: f3f68fd5 radio: 1.15.0
4468 [BKD] Current device changed to "H0RU"
4568 [RPC] Starting RPC session...
4659 [RPC] RPC session started successfully.
4659 [RPC] (1) Gui Start ScreenStream START
4717 [RPC] (1) Gui Start ScreenStream SUCCESS
18645 [DEV] Full Update @H0RU START
18648 [RPC] (2) Gui Start VirtualDisplay START
18650 [UTL] Region Provisioning @H0RU START
18650 [DBG] Development hardware region detected, skipping region provisioning...
18650 [UTL] Region Provisioning @H0RU SUCCESS
18650 [UTL] Refresh Storage Info @H0RU START
18706 [RPC] (2) Gui Start VirtualDisplay SUCCESS
18706 [RPC] (3) Storage Info @/ext START
18760 [RPC] (3) Storage Info @/ext SUCCESS
18761 [RPC] (4) Storage Stat @/ext/Manifest START
18825 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
18835 [UTL] Refresh Storage Info @H0RU SUCCESS
18837 [ZIP] Uncompressing file with size of 3347830 bytes...
18887 [ZIP] Uncompression finished : No Error
18901 [UTL] Create Path @/ext/update/f7-update-unlshd-064e START
18904 [RPC] (5) Storage Stat @/ext/update START
18963 [RPC] (5) Storage Stat @/ext/update SUCCESS
18964 [RPC] (6) Storage Stat @/ext/update/f7-update-unlshd-064e START
19032 [UTL] Create Path @/ext/update/f7-update-unlshd-064e SUCCESS
19032 [RPC] (6) Storage Stat @/ext/update/f7-update-unlshd-064e SUCCESS
19033 [UTL] Verify checksum of 6 items @H0RU START
19033 [RPC] (7) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/firmware.dfu START
20363 [DBG] File is identical: "/ext/update/f7-update-unlshd-064e/firmware.dfu"
20363 [RPC] (7) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/firmware.dfu SUCCESS
20363 [RPC] (8) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/radio.bin START
20630 [DBG] File is identical: "/ext/update/f7-update-unlshd-064e/radio.bin"
20631 [RPC] (8) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/radio.bin SUCCESS
20631 [RPC] (9) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/resources.tar START
20710 [DBG] File changed: "/ext/update/f7-update-unlshd-064e/resources.tar" old: "496fbb35b2059373a0a30d5acfe7376a" new: "f8bb8fb22fd5df7d2b78a56d512cf8b8"
20710 [RPC] (9) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/resources.tar SUCCESS
20710 [RPC] (10) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/splash.bin START
20773 [DBG] File does not exist: "/ext/update/f7-update-unlshd-064e/splash.bin"
20774 [RPC] (10) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/splash.bin SUCCESS
20774 [RPC] (11) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/update.fuf START
20837 [DBG] File does not exist: "/ext/update/f7-update-unlshd-064e/update.fuf"
20837 [RPC] (11) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/update.fuf SUCCESS
20837 [RPC] (12) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/updater.bin START
20900 [DBG] File does not exist: "/ext/update/f7-update-unlshd-064e/updater.bin"
20900 [RPC] (12) Storage Md5Sum @/ext/update/f7-update-unlshd-064e/updater.bin SUCCESS
20900 [UTL] Verify checksum of 6 items @H0RU SUCCESS
20900 [UTL] Upload 4 entries @H0RU START
20901 [RPC] (13) Storage Write @/ext/update/f7-update-unlshd-064e/resources.tar START
24637 [RPC] (13) Storage Write @/ext/update/f7-update-unlshd-064e/resources.tar SUCCESS
24640 [RPC] (14) Storage Write @/ext/update/f7-update-unlshd-064e/splash.bin START
54632 [UTL] Upload 4 entries @H0RU ERROR: Operation timeout (generic)
54632 [DEV] Full Update @H0RU ERROR: Operation timeout (generic)
54632 [BKD] Current operation finished with error: "Operation timeout (generic)"
54633 [RPC] (14) Storage Write @/ext/update/f7-update-unlshd-064e/splash.bin ERROR: Operation timeout (generic)
```

RogueMaster, 2023-10-11:
You may have a faulty SD card

LuizizM, 2023-10-11:
Found out after debuging it was the SD card was full due to me forgetting to delete the previous update packages on the updates folder. Closing the issue.

closed 2023-10-11 by LuizizM