DarkFlippers/unleashed-firmware #627: TPMS modulation

DavidIgor, 2023-10-11:
when Protoview is useless for replaying tpms signals (as it cripples the sensor ID and data), please someone provide frequency modulation presets that I could just paste to that setting file and be able to record and replay tpms signals successfully. 

thanks

haisenteck, 2023-11-22:
this work on my firmware... but i hardcoded in the library of subghz.
```
const uint8_t subghz_device_cc1101_preset_TPMS[] = {
	0x02, 0x0D, 0x07, 0x04, 0x08, 0x32, 0x0B, 0x06, 0x10, 0x69, 0x11, 0x93, 0x12, 0x14, 0x13, 0x01, 0x14, 0xF8, 0x15, 0x40, 0x18, 0x18, 0x19, 0x16, 0x1B, 0x07, 0x1C, 0x00, 0x1D, 0x91, 0x20, 0xFB, 0x21, 0x56, 0x22, 0x10, 0x00, 0x00, 0xC0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
```

DavidIgor, 2023-11-22:
@haisenteck  hi , in settings_user ? 
thanks

haisenteck, 2023-11-22:
> @haisenteck  hi , in settings_user ? 
> thanks

No, the code that i give there Is for the lib of subghz. Is ti adapt for the settings.

DavidIgor, 2023-11-22:
> > @haisenteck  hi , in settings_user ? 
> > thanks
> 
> No, the code that i give there Is for the lib of subghz. Is ti adapt for the settings.

Can you please translate it so that we can just paste it into setting-user ? 
 or what needs to be done in order to do so?