DarkFlippers/unleashed-firmware #629: Add Motorline garage door remotes

Yomtus, 2023-10-12:
### Description of the feature you're suggesting.

Can someone add Motorline garage door remotes to the rolling codes?
[image]

Yomtus, 2023-10-17:
I can offer all kinds of information about that model of control and the motor plate. Since it is the one I have in my house.

xMasterX, 2024-06-14:
Can you provide RAW recordings of this remote?

Yomtus, 2024-06-14:
Hello, here i send a raw file with 20 records of a Motorline control, thanks for colaboration.

[RAW_MOTORLINE_X20.zip](https://github.com/user-attachments/files/15839243/RAW_MOTORLINE_X20.zip)

CombatMode, 2025-04-27:
I have this MFK, it uses KeeLoq Normal learning and 10 bits discriminator.

[image]
[image]

xMasterX, 2025-04-27:
> I have this MFK, it uses KeeLoq Normal learning and 10 bits discriminator.
> 
> [image] [image]

So, maybe you can share the key? Contact me in discord or telegram so we can add that to the firmware

CombatMode, 2025-04-28:
@xMasterX 
I need to discuss this with my partner first. We are very interested in moving "heavy" code (like NFC, SubGHz, etc) into external resources to free as much internal flash as possible. We have more new keys besides this and would like to cooperate.

xMasterX, 2025-04-28:
> [@xMasterX](https://github.com/xMasterX) I need to discuss this with my partner first. We are very interested in moving "heavy" code (like NFC, SubGHz, etc) into external resources to free as much internal flash as possible. We have more new keys besides this and would like to cooperate.

Ok, waiting for updates, if you can share mfkeys that we can add to firmware that would be good

CombatMode, 2025-04-29:
> Ok, waiting for updates, if you can share mfkeys that we can add to firmware that would be good

Good news, my partner agreed to provide mfkey, but only on one condition: if it is possible to move heavy code to SD card resources to free at least 30 kilobytes more of internal flash (the more, the better, of course) w/o losing existing functionality. In this case, we all win: everyone will have more space for internal code and new protocols.

For example, in the latest build (with NFC changes) there is about 300 bytes less free int flash and this makes us sad.

xMasterX, 2025-04-30:
> Good news, my partner agreed to provide mfkey, but only on one condition: if it is possible to move heavy code to SD card resources to free at least 30 kilobytes more of internal flash (the more, [...]

See, one problem here, If I didn't know who you are, thats would sound like good idea, sadly for you I knew who's behind that empty new account just after the first message with photos and text about that motorline with 10bits, each new message just approved that I was right, so, whats in result
I'm glad to hear that Derrow agreed to share his key (that key is already shared with you)

About your "conditions", If you want to actually help the project and our team with something, there's always was a way, it wasn't hard to make that account, nor its not that hard to send a PR with some fixes, or share the keys, or something to our team members in telegram/discord if you want to be anonymous, any actual help will be accepted, this is opensource project
There's cannot be any "conditions", exchanges, etc, in opensource project, thats not how it works, If you dislike someone from the team this is not a C code problem, PRs with non malicious stuff are reviewed equally for everyone, this is not a linux core where you need to not have specific passport.

But basically you are trying to exchange other guy's mfkey to the work that would in result help you doing your criminal stuff and continue spamming our users with that
Thats not how it should work, and not how it works

I can tell our conditions then:
Stop spamming our telegram chat users in DMs, then do something actually good for the project if you have interest in that, 
then we talk, then we discuss, maybe we can even pretend that we didn't seen your "copypastas" and insults about me and your criminal car stealing stuff exist if you leave our project, users and team in peace and stop using our name for your things, as for now we are not doing any work for you, not for money, not for mfkeys, its not worth it

CombatMode, 2025-04-30:
@xMasterX 

I see you are confusing me with that guy who was in your team and became persona non grata. I'd say I know him and even used to contact. But I don't share his interests and we're not working together. I am working at my own project, also Unleashed-based, but with Sub-GHz automations main goal. 

What about internal flash - this is not a work for someone personally, but for everyone. It only a matter of time: soon or later everyone faces this limitation and will have no other options - only to move code to external resources.

I just offered you the information that I suppose no one else will. For the good job that everyone needs. But you met this with angry point.

li0ard, 2025-07-09:
Im also have this MF. I can give it for free without any "conditions".

[image]

[image]

> I just offered you the information that I suppose no one else will. For the good job that everyone needs. But you met this with angry point.

At the moment, I (like most people, in my opinion) don't have any problems with internal flash. The project is open, and you can freely modify it and send PRs.

xMasterX, 2025-07-26:
@Yomtus
Please verify at latest release, let us know if you got it working, also will be great to check if new remote created via Add manually menu will bind to original receiver.

closed 2025-07-26 by xMasterX

Yomtus, 2025-07-27:
It works perfectly. Thank you very much.